dl9rdz/rdz_ttgo_sonde #95: No GPS lock showing, even with GPS LED blinking.

sgtwilko, 2021-06-29:
**Describe the bug**
TTGO T-Beam V1 with built-in GPS does not show a GPS location even when the GPS LED starts to flash. 

**To Reproduce**
Steps to reproduce the behaviour:
I used a board that I had used for meshtastic and installed the latest Dev firmware.
Issue presented!

**Expected behaviour**
GPS lock is shown on screen type 4.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version and specific configuration:**
 - devel20210619-B11-full.bin (Jun 27 17:01:33 2021)
 - Standard TTGO TBeam V1 with small OLED.
 
**Additional context**
I've had to reset the GPS with the sketch linked in the the first comment of issue #44 

First time it booted I saw the message ```GPS factory reset...```

However it didn't show a GPS lock even when the GPS LED was blinking, and I also see the following in the log:
```GPS reset file: not found/isdir```

Not sure if the latter is only subsequent boots, I didn't see it on the first boot, but I wasn't looking.

dl9rdz, 2021-06-30:
Thanks for the report. I guess I should modify this as this frequently causes confusion.

My current GPS factory reset code is minimalistic. They only thing it does is to reset the persistent GPS config to factory defaults.  This is done only once after flashing the firmeware, so the observed behaviour (printing factory reset on the first time, when it deletes a file on flash, and then its normal behaviour that the file does not exist any more on next reboot).

However, the GPS chip uses the persistent GPS config as its config only after it is power cycled. A simple reset is not enough, you have to remove the USB cable and the 16550 battery. At least this is what I have successfully tested here with meshtastic installed before flashing my firmware.  So maybe you the battery in the TTGO all the time (or a USB cable continuously connected)?

sgtwilko, 2021-06-30:
Hiya.

I had power cycled, and I have no battery in this TBeam, it was only after that other sketch had done the magic that the OLED showed the GPS lock.

That other sketch seems to cycle through several different serial speeds before resetting to 9600 and factory resetting the GPS, but don't know how thorough you need to be!

It would also be handy if the serial console showed more debug regarding the GPS, I couldn't see anything even confirming if it had even started talking to the GPS or if it had lock.

gariac, 2021-07-19:
I have a ttbeam 1.1. The GPS doesn't work. I am not really set up to compile code or I would need a good guide. Technically cross compile. If #44 is a solution could someone provide a binary?

closed 2025-11-21 by dl9rdz