dl9rdz/rdz_ttgo_sonde #96: WiFi in mode 3 fails to connect to AP, but works in mode 1

sgtwilko, 2021-06-29:
**Describe the bug**
When left at the default Wifi setting of 3, it attempts to connect to my AP but gives up very quickly and starts the built in AP.
If I change it to mode 1 it will connect fine.
It also connected fine in mode 3 on the 0.8.0 firmware.

**To Reproduce**
Steps to reproduce the behaviour:
Use my AP and try to connect in mode 3

**Expected behaviour**
It connects to the configured AP.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version and specific configuration:**
 - devel20210619-B11-full.bin (Jun 27 17:01:33 2021)
 - TTGO TBeam v1 standard small OLED.

**Additional context**
worked fine on 0.8.0 doesn't work on latest dev build.

Jan-NiklasB, 2021-07-16:
Can also confirm this issue with "devel20210630-B11-full.bin" flashed on a T-Beam Lora32 T22 V1.1.

Seems like the ESP can't hold enough SSID's and if the wanted AP isn't one of the first found the ESP seems to stop scanning and opens it's own AP.

Switching to AP Mode 1 solves the problem temporarily, but if the AP isn't present, there's no way to reconfigure the TTGO.

LukePrior, 2021-07-28:
I am encountering this same issue on devel20210726, here is my log:

Network MYWIFI: RSSI -88, MAC MACMACMAC, enc: WPA2_PSK
Match for MYWIFI at 2
Match found at scan entry 1, config network 2
Network MYWIFI: RSSI -91, MAC MACMACMAC, enc: WPA2_PSK
Match for MYWIFI at 2
Connecting to: MYWIFI with password PASSWORD
......[WiFi-event] event: 3
WiFi clients stopped
[WiFi-event] event: 0
WiFi interface ready
[WiFi-event] event: 2
WiFi client started
Reconnecting to: MYWIFI with password PASSWORD
..........[WiFi-event] event: 4
Connected to access point
....[WiFi-event] event: 3
WiFi clients stopped
[WiFi-event] event: 3
WiFi clients stopped
Activating access point mode
[WiFi-event] event: 0
WiFi interface ready
Wait 100 ms for AP_START...
[WiFi-event] event: 14
WiFi access point started
Ready
updateDisplayIP: 5 7
AP IP address: 192.168.4.1

dl9rdz, 2021-07-28:
Is this also a problem on the most recent version?

Looks like a timing issue. It takes 5s to connect (echo "." is 500ms), and then after 2s more (7s in total) Mode 3 gives up (as it has not yet received an IP via DHCP after those 7 sec.. In the most recent version, this timeout should now be 10s in total instead of 7s (I removed the "Reconnecting" part which was needed in the past due to some bug in the wifi libraray, which seems no longer to be present). 

If that still causes issues, one could also use separate time outs for connect and DHCP.

LukePrior, 2021-07-28:
> Is this also a problem on the most recent version?

I have updated to 20210728 and the issue now seems to appear about half the time.

Here is the updated log:

```
Network MYWIFI: RSSI -85, MAC MACMACMAC, enc: WPA2_PSK
Match for MYWIFI at 2
Match found at scan entry 1, config network 2
Connecting to: MYWIFI with password PASSWORD
..........[WiFi-event] event: 5
Disconnected from WiFi access point
.........[WiFi-event] event: 5
Disconnected from WiFi access point
.[WiFi-event] event: 3
WiFi clients stopped
Activating access point mode
[WiFi-event] event: 0
WiFi interface ready
Wait 100 ms for AP_START...
[WiFi-event] event: 14
WiFi access point started
Ready
updateDisplayIP: 5 7
AP IP address: 192.168.4.1
```

Jan-NiklasB, 2021-07-28:
Well, I tried a bit around with the settings and found out that the behaviour is more or less inconsistent.

After initial problems I set the mode to 1, but later again to 3 when going on chase.
But after initial setting it to 1 it seems to work quite fine...

So maybe a problem with the initial config shipped with the bin file?

dl9rdz, 2021-08-06:
The connect + DHCP timeout in mode 3 is now twice as long (20s instead of 10s). Should be sufficient for even slow APs / DHCP servers. If not, use mode 1.

closed 2021-08-06 by dl9rdz