dl9rdz/rdz_ttgo_sonde #97: Feedback on using aprs.fi's IOS App with rdz_ttgo_sonde

HB9FUF, 2021-07-04:
Hi DL9RDZ & Co,

Thanks for this great app!  This is not as such a bug report, not a change suggestion, merely a feedback report on trying to use _aprs.fi_'s IOS App ( https://apps.apple.com/us/app/aprs-fi/id922155038 ) with _rdz_ttgo_sonde_.

1. The app looks up TCP KISS TNCs via mDNS, looking for the service "_kiss-tnc._tcp"; note the hyphen between "kiss" and "tnc".  Probably according to this spec: https://github.com/hessu/aprs-specs/blob/master/TCP-KISS-DNS-SD.md
2. The KISS data frames are not accepted by the app unless a \x00 byte is inserted right after the frame-start marker \xC0.  It *seems* to be part of the KISS spec (0 for port-0, and 0 for KISS Frame Type = Data). Same problem with WB2OSZ's _kissutil_ and KA2DDO's _YAAC_.
3. The app reports the APRS frames with timestamps *HHMMSSz as error/warning invalid timestamps, and using the postfix "h" instead of "z" addresses it.
4. For some reason, the mDNS registrations from the ESP32 (TTGO T-Beam V1.1) do not seem to reach my IOS iPad, both WiFi connected to the same home WiFi AP (tested with IOS app _Flame Services Browser_, https://apps.apple.com/us/app/flame-services-browser/id325206381 and verified iPad could "see" the ESP32's IP & port with IOS App _Fing_, https://apps.apple.com/us/app/fing-network-scanner/id430921107 ). Using the ESP32 in AP mode, and connecting the iPad to that ESP32's AP works around the problem. But as I am however able see on the iPad the mDNS registrations issued, from my same-subnet LAN-wired Windows machine, with the command `dns-sd -R RDZKISSTNC _kiss-tnc._tcp local 14580`, I can simply establish a TCP tunnel from the PC to the ESP32 with NMap.org's ncat ( https://nmap.org/ncat/ ) with `ncat --listen IP.Addr.Win.PC 14580 --sh-exec "ncat IP.Addr.RDZ.TTGO 14580"`

With the above 4 "tricks", _aprs.fi_ IOS app works really good with _rdz_ttgo_sonde_.

I am amazed to see a nearly on-part decoding-sensibility between my _Airspy-R2_ + _IW1GIS's RS41 Tracker_ ( http://escursioni.altervista.org/Radiosonde/ ) and my _rdz_ttgo_sonde_ when both attached to my home vertical 70cm antenna via a 2-way splitter!

Thanks again for this great app & 73s
Pierre HB9FUF

DL2MF, 2021-08-17:
There are several other platforms, focussed on visualizing radiosonde data, APRS is not the right choice, only because other people also using this. Most of the content is not amateur radido specific, wrong formatted, using invalid callsing or ignoring the APRS protocol specifications (http://www.aprs.org/doc/APRS101.PDF and addendum http://www.aprs.org/aprs12.html).

dl9rdz, 2021-08-17:
Hi, while I agree that aprs is not ideal for radiosonde tracking, it exists and the points you make are really (minor) bugs that should be fixed. APRSdroid dos not care about (2) and (3), and mDNS support is still an open feature requests. The latest commit still works fine with APRSdroid, and is likely to work with your iSO app as well. I changed the mDNS part as well, and checked it with dns-sd command line, so it *should* be fine, but it is not supported yet by APRSdroid anyway, and I don't have an iOS device for testing...

HB9FUF, 2021-08-24:
Hi DL9RDZ,

The changes you made in commit [b69a726](https://github.com/dl9rdz/rdz_ttgo_sonde/commit/b69a726df74d257c2cc8d5353110976318661026) are the exact same changes I had made in my copy of your repo early July.  That would definitely fix it with aprs.fi's IOS App.  As far as I am concerned, this issue should now be closed.

Thank you much!
Pierre HB9FUF

dl9rdz, 2021-09-13:
Thanks, closing this issue.

closed 2021-09-13 by dl9rdz