dl9rdz/rdz_ttgo_sonde #98: upload to Radiosondy and SondeHub does not work

hem0773, 2021-07-11:
upload to Radiosondy and SondeHub does not work

Radiosondy here I entered the data from the account
Complete all fields at SondeHub

Do I need a GPS FIX?

LukePrior, 2021-07-11:
> upload to Radiosondy and SondeHub does not work
> 
> Radiosondy here I entered the data from the account
> Complete all fields at SondeHub
> 
> Do I need a GPS FIX?

Can you send a picture of your SondeHub configuration settings?

hem0773, 2021-07-12:
> 
> 
> > upload to Radiosondy and SondeHub does not work
> > Radiosondy here I entered the data from the account
> > Complete all fields at SondeHub
> > Do I need a GPS FIX?
> 
> Can you send a picture of your SondeHub configuration settings?

hem0773, 2021-07-12:
> 
> 
> > upload to Radiosondy and SondeHub does not work
> > Radiosondy here I entered the data from the account
> > Complete all fields at SondeHub
> > Do I need a GPS FIX?
> 
> Can you send a picture of your SondeHub configuration settings?
[image]

LukePrior, 2021-07-12:
> > > upload to Radiosondy and SondeHub does not work
> > > Radiosondy here I entered the data from the account
> > > Complete all fields at SondeHub
> > > Do I need a GPS FIX?
> > 
> > 
> > Can you send a picture of your SondeHub configuration settings?
> > [image]

Set chase location active to 0 (you have it enabled so it is searching for hardware GPS which it cannot find).
You should also set the antenna field to a string such as "dipole antenna"

hem0773, 2021-07-12:
> Set chase location active to 0 (you have it enabled so it is searching for hardware GPS which it cannot find).
> You should also set the antenna field to a string such as "dipole antenna"

sondehub.org

hem0773, 2021-07-12:
> Set chase location active to 0 (you have it enabled so it is searching for hardware GPS which it cannot find).
> You should also set the antenna field to a string such as "dipole antenna"

made the changes!

No success, nothing is displayed on sondehub.org

hem0773, 2021-07-12:
[image]
THX

closed 2021-07-12 by hem0773

Jan-NiklasB, 2021-07-22:
@LukePrior 
So it I understand the posts Herr right, the chasemode isn't functional AG the Moment, right?

It thats the Gase, it would be cool if the TTGO would report it's position when opening sondehub in the web UI.

Greetings
Jan-Niklas

LukePrior, 2021-07-22:
> So if I understand the posts here right, the chasemode isn't functional at the Moment, right?

Chase mode works if you have a GPS connected to the TTGO device

Jan-NiklasB, 2021-07-22:
Well, I have a T-beam including a Neo6m.
But it I try to set it up and open Sondehub over the UI I get a json response with forbidden. 

If you want me to, I can open a separate issue for this.

LukePrior, 2021-07-22:
> Well, I have a T-beam including a Neo6m.
> But it I try to set it up and open Sondehub over the UI I get a json response with forbidden.
> 
> If you want me to, I can open a separate issue for this.

Yeah I'm not entirely sure about that might be a good idea to open a new issue and post all the details.

Jan-NiklasB, 2021-07-22:
I can send you my config in this thread tomorrow, maybe you can tell me directly where my issue comes from or that I should open a new issue.