dl9rdz/rdz_ttgo_sonde #99: TTGO and Sondehub

14ri004, 2021-07-20:
hello, little question, with the M10 nothing is displayed on probehub, no callsign, nor the decoding data, because the decoding of the RS41 is sent to Sondehub, and works perfectly, it is a software bug or it comes from my home ??

thank you

LukePrior, 2021-07-20:
Hi,

I can have a look into this what is your SondeHub station callsign?

14ri004, 2021-07-20:
hello, callsign RDZ_TTGO it works with RS41s, but not with M10s, so it's not a problem of callsign

LukePrior, 2021-07-20:
I can check if data from your device is being uploaded to the SondeHub backend. You can also check if the program is displaying any error messages by checking the serial log while the device is detecting a M10 sonde.

LukePrior, 2021-07-21:
Ok I've had a look and it seems like the data isn't making it to the SondeHub backend so it must be an error with the program or one of the integrated checks failing.

You can see all the data uploaded from your station on the last 3 days here (Google sign in required) https://es.v2.sondehub.org/_plugin/kibana/app/discover#/?_g=(filters:!(),refreshInterval:(pause:!t,value:0),time:(from:now-3d,to:now))&_a=(columns:!(_source),filters:!(('$state':(store:appState),meta:(alias:!n,disabled:!f,index:'8fec99f0-6222-11eb-b862-2f0a42a5ce55',key:uploader_callsign,negate:!f,params:(query:RDZ_TTGO),type:phrase),query:(match_phrase:(uploader_callsign:RDZ_TTGO)))),index:'8fec99f0-6222-11eb-b862-2f0a42a5ce55',interval:auto,query:(language:kuery,query:''),sort:!())

You will need to provide serial logs from when the device is decoding a M10 sonde.

14ri004, 2021-07-21:
thank you for your answer, there are sometimes errors on the TTGO screen, this could be due to the setting of the reception of the probes ?, I do not yet know this TTGO software very well, but other people use the TTGO and arrive has decoded the M10, I will eventually get there.

LukePrior, 2021-07-21:
> thank you for your answer, there are sometimes errors on the TTGO screen, this could be due to the setting of the reception of the probes ?, I do not yet know this TTGO software very well, but other people use the TTGO and arrive has decoded the M10, I will eventually get there.

I can't help with the onscreen errors but to get the serial logs you will need to attach the TTGO to your computer and view the serial logs using a tool like the Arduino IDE. This will show all the log messages from the device and could potentially diagnose the issue.

14ri004, 2021-07-21:
holala this is not winning this story, I will find out why it does not work for me, thank you for your help

14ri004, 2021-07-23:
hello, it still does not work, impossible to see the M10 probes on SondeHub, while I decode the RS41 perfectly and they are well displayed, when I decode an M10 I have a lot of errors so not possible to decode and see the M10s on SondeHub, I use another TTGO which is exactly the same,

closed 2021-07-25 by 14ri004