dl9rdz/rdz_ttgo_sonde #100: M10 undecodable

14ri004, 2021-07-22:
Hello, need help or explanations thank you, I can't decode the M10s or with a lot of errors, with the RS41 the decoding is done perfectly, but the M10 is barely 10%, not possible to follow it, would someone have the right settings, I spent the afternoon changing the settings but that didn't change anything, I only have 1 frame or 2 from time to time.

here are my settings:
M10 / M20 AGC bandwidth 12500
M10 / M20 RX bandwidth 25000

thank you so much

14ri004, 2021-07-24:
hello, still no detection of M10 with TTGO T BEAM v1.1
no frames, no gps see screenshot, unable to send information to SondeHub

[image]

Mooni123, 2021-07-26:
Hello,

the shown 'Frame# 0, Sats=0' in the WebInterface, is normal for M10/M20 radiosondes.
These sondes have no framecounter in their data, so other decoder will generate a virtual framenumber
from GPS time for the Sondehub database upload. This framenumber is not displayed in LCD or WebUI.
The 0 sats is also normal as this is not decoded from sonde data at the moment.

About your reception errors, I think your signal is too weak for proper decoding.

14ri004, 2021-07-26:
Thank you for your answer, if I understand correctly, the M10s are not taken into account on SondeHub, I managed to decode an M10 but no follow-up on SondeHub, well too bad, thank you.Envoyé depuis mon appareil Galaxy
-------- Message d'origine --------De : Mooni123 ***@***.***> Date : 26/07/2021  17:30  (GMT+01:00) À : dl9rdz/rdz_ttgo_sonde ***@***.***> Cc : 14ri004 ***@***.***>, Author ***@***.***> Objet : Re: [dl9rdz/rdz_ttgo_sonde] M10 undecodable (#100) 
Hello,
the shown 'Frame# 0, Sats=0' in the WebInterface, is normal for M10/M20 radiosondes.
These sondes have no framecounter in their data, so other decoder will generate a virtual framenumber
from GPS time for the Sondehub database upload. This framenumber is not displayed in LCD or WebUI.
The 0 sats is also normal as this is not decoded from sonde data at the moment.
About your reception errors, I think your signal is too weak for proper decoding.

Mooni123, 2021-07-26:
No, that was only the reason why you have  'Frame# 0, Sats=0' on WebUI !!

Data Upload to Sondehub should work, but this part was added by LukePrior
and I never tried that.

There were some fixes in last develop build the last days
Maybe you can try this, if you don't have it already.

14ri004, 2021-07-26:
I updated my 2 ttgo with the last version of rdz, with the rs41 no worries SondeHub works perfectly, but with an M10 nothing, no monitoring of the M10s, strange that it works with an rs41 but not with an M10 ?? ?

dl9rdz, 2021-07-26:
OK, so the sondehub code did not upload data with sats==0.

For M10, I have now added code to decode the #sats. I tested this with M10 810-2-13028 and it seem to work find.
For M20, there is no known #sats in the data (at least dxlaprs and autorx both do not decode that). I now changed the code to upload data to sondehub if sats==0 and type is M20.

As for the decode quality, the default RX bandwith for M10 is probably not ideal. I rarely receive M10 here, so no chance to optimize that easily with live signals. Try different larger or smaller values and observe if decode quality improves. Probably you need larger values.

14ri004, 2021-07-26:
thank you for the modification, I will test and let you know if it works on my side, thank you

14ri004, 2021-07-27:
hello, indeed for the M10, it works perfectly now, good job, thank you

closed 2021-07-27 by 14ri004