dl9rdz/rdz_ttgo_sonde #101: Problems with Sondehub API

Jan-NiklasB, 2021-07-23:
**Describe the bug**
If not setting the Position for sondehub the upload oft the data received doesn't seems to work, it setting a position I get Access denied. 

**To Reproduce**
Run the devel bin from mid june on a T-Beam V1.1 and try to set it up for sondehub. 

**Expected behavior**
Station or chase car appears on sondehub 

**Screenshots**
[image]
This config doesn't seem to do anything at all

[image]
This config produces the error in the next screenshot

[image]

**Version and specific configuration:**
 - Devel bin from mid june (17th june if I remember right, I don't have the T-Beam here right now)
 - TTGO T-Beam V1.1

**Additional context**
Problem was also mentioned in this issue #98 with @LukePrior

LukePrior, 2021-07-23:
Ok so going off your configuration settings it looks like in the first image data should be correctly uploaded to SondeHub but your station won't be visible as you haven't set your position or enabled chase car mode.

The second configuration looks correct and I can confirm that data has been successfully uploaded to SondeHub on the 22nd July. You uploaded 190 data packets to the SondeHub database and your station should have been visible on sondehub.org.

The third image shows a relatively new feature where you can access the SondeHub tracker directly from the web interface, this is exactly the same as if you were to navigate to sondehub.org from any other device. We recently updated the tracker which appears to have broken the links used in this program so I have opened #102 to resolve that. When the code has been merged that error should disappear.

TLDR: Your second configuration is working correctly, the integrated tracker is currently broken but you can access it from any device at sondehub.org

Jan-NiklasB, 2021-07-23:
Thanks for the fast reply.

Could you may explain me or give me a hint on how to check it data was uploaded?

Currently I set it to chase mode but can't see it on sondehub (but I also don't receive a sonde at the moment if thats a requirement )

LukePrior, 2021-07-23:
> Could you may explain me or give me a hint on how to check it data was uploaded?

You can view the raw data in the Elastic Search [page](https://es.v2.sondehub.org/_plugin/kibana/app/home#/). You can use this [link](https://es.v2.sondehub.org/_plugin/kibana/app/discover#/?_g=(filters:!(),refreshInterval:(pause:!t,value:0),time:(from:now-2w,to:now))&_a=(columns:!(_source),filters:!(('$state':(store:appState),meta:(alias:!n,disabled:!f,index:'8fec99f0-6222-11eb-b862-2f0a42a5ce55',key:uploader_callsign,negate:!f,params:(query:BOGATKAJ),type:phrase),query:(match_phrase:(uploader_callsign:BOGATKAJ)))),index:'8fec99f0-6222-11eb-b862-2f0a42a5ce55',interval:auto,query:(language:kuery,query:''),sort:!())) to see the data uploaded by your device in the last two weeks. You will need to create an account to view these pages.

> Currently I set it to chase mode but can't see it on sondehub (but I also don't receive a sonde at the moment if thats a requirement )

Hmm, so if you have Sondehub chase location active set to 1 and have a GPS lock on your device when it is powered on it should upload its position to SondeHub. This should happen every 30 seconds so I would check the serial logs by attaching the device to your computer and checking the logs.

hem0773, 2021-07-23:
[image]
with the setting and GPS fix, my receiver is not visible

[image]
with the setting it goes to save and reboot directly

Board V1.1 Perhaps this is the problem

Why not write the data into the table in GPS Fix where you entered the data by hand?

Jan-NiklasB, 2021-07-23:
Well, station mode with position set by hand works, but if clearing the position and enabling chase mode, i'm not visible on the map and there also seems to be no data transmitted

LukePrior, 2021-07-23:
> with the setting and GPS fix, my receiver is not visible

You will only appear on map of you are decoding a sonde and have GPS fix, is this the case?

LukePrior, 2021-07-23:
> but if clearing the position and enabling chase mode, i'm not visible on the map and there also seems to be no data transmitted

Once again are you decoding any sonde when this is happening?

If so I will need to see the serial logs to help any further.

LukePrior, 2021-07-24:
> with the setting and GPS fix, my receiver is not visible

> Well, station mode with position set by hand works, but if clearing the position and enabling chase mode, i'm not visible on the map

I have found the issues causing this and have solved them in #104 

The station position can now be uploaded even if no sonde is detected and the request format has been modified so it doesn't fail.

LukePrior, 2021-07-24:
You will need to wait for the changes to be merged and a new image created.

Jan-NiklasB, 2021-07-24:
> You will need to wait for the changes to be merged and a new image created.

Cool, thanks for your effort!

I think I have to download the code and familiarize myself with it...

LukePrior, 2021-07-24:
Ok so the map issue should be fixed I will update the other pull request to bump the version number so it is easier to merge.

LukePrior, 2021-07-27:
Can you update the latest devel image and check if these issues have been resolved?

Jan-NiklasB, 2021-07-27:
Sure, i will try it when i get home today

LukePrior, 2021-07-27:
> Sure, i will try it when i get home today

I stuffed up so it won't work until this is merged: https://github.com/dl9rdz/rdz_ttgo_sonde/pull/113

Jan-NiklasB, 2021-07-27:
> > Sure, i will try it when i get home today
> 
> I stuffed up so it won't work until this is merged: #113

It's merged and works as it's supposed to do.
Nice work!
Thanks!

closed 2021-07-27 by Jan-NiklasB