dl9rdz/rdz_ttgo_sonde #106: Probe number

14ri004, 2021-07-24:
I just did a test with a probe, and I noticed that the number of the probe does not correspond to this one?

906211077 and on RDZsonde M9622435 ??

I updated the last devel: 20210630-B11-full.bin (Jul 24 14:57:03 2021)

[image]
[image]

dl9rdz, 2021-07-24:
rdzsonde uses two id formats, as you can see in the wifi tab
ser is the serial as printed on the sonde (at least for most types where this is possible)
id is the object id used on aprs.fi (I try to keep this in sync with dxlaprs and autorx)

If you want to see the full serial number on the TFT, you have to edit screens2.txt yourself, replace all instances of "`=I`" or "`=Is`" by "`=In`".  (Not tested, no guarantee that the longer serial number will fit in the screen, as internally the serial number is represented with separators between the three blocks, as seen on the web interface.

14ri004, 2021-07-24:
Thank you for your quick reply,

closed 2021-07-24 by 14ri004