dl9rdz/rdz_ttgo_sonde #107: Cache SondeHub Uploads

LukePrior, 2021-07-25:
We currently upload every single frame to SondeHub as an individual request which is extensive on the SondeHub servers.

This creates a large number of requests on the back-end and will become a larger issues as more devices join the network.

We currently handle this in radiosonde_ato_rx by caching the last 15s of data and uploading as a single request.

I'm not sure if this approach would be possible to implement but I did get a suggestion from Michaela Wheeler that we could leave the HTTP connection open for several packets as CloudFront has a 30s timeout.

I'm not too sure how best to implement that idea so if anyone wants to give it a go it would be much appreciated.

TheSkorm, 2021-07-25:
Even buffering 3-5 packets would make a huge difference :)

dl9rdz, 2021-07-27:
Last commit is a quick attempt. Just briefly tested

LukePrior, 2021-07-28:
Awesome that seems to work ok now.

closed 2021-07-28 by LukePrior