dl9rdz/rdz_ttgo_sonde #118: update.ino.bin file ????

14ri004, 2021-07-28:
hello, what is the update.ino.bin file for ??, I tried to update but the TTGO refuses to start, forcing to install the full version, but which unfortunately erases all my information, I have to redo all my configuration (freq, wifi, conf etc ...) with each new version of RDZ devel, thank you for your work.

LukePrior, 2021-07-28:
You can use the OTA update feature at YOUR_DEVICE_IP/update.html

dl9rdz, 2021-07-28:
http://rdzsonde.mooo.com/:

"If you want to update only the software, you can flash update.ino.bin at offset 0x10000 (or use OTA update via http://rdzsonde.local/update.html)."

Note the 0x10000 instead of 0x1000. The full image contains the partition table (at 0x1000), the software (at 0x10000) and the file system (whereever that is). The update is only the software part.

LukePrior, 2021-07-28:
> or use OTA update via http://rdzsonde.local/update.html

How does this work as when I try the link on my local network with TTGO connected it doesn't find the device and it need to navigate to the specific IP.

dl9rdz, 2021-07-28:
It works if your computer (with the web brower) and the TTGO are on the same subnet and if the computer supports mdns discovery. Its fine on MacOS and iOS, other OS and browsers it may or may not work. If its not working, you have to manually use the IP address instead.

14ri004, 2021-07-28:
impossible to do the prog, unless I do a starp between GPIO0 and GND, on all my TTGO it's exactly the same, it works but with the strap,

14ri004, 2021-07-28:
actually with 0x10000 no deletion of my config, perfect update, thank you for your help.

closed 2021-07-29 by 14ri004