dl9rdz/rdz_ttgo_sonde #122: Include link to update page in main interface?

LukePrior, 2021-08-07:
Hello,

What do you think about adding a link to the OTA update page /update.html from the About tab in the main interface?

I think this would make it easier for new people to find and what would be even better is if we check what the latest version is if we have an internet connection and display some update available text next to the current version number?

[image]

Cheers

dl9rdz, 2021-08-07:
Ok, I did a first step in this direction.

Travis-CI creates a HTML file with version info of the update on rdzsonde.mooo.com (== gh-pages on github)

I currently just include this as iframe on update.html. So the TTGO has no idea what the update version is, just the user's browser fetches that info. And the about tab has a link to update.html.

I guess that's fine for now. More sophisticated UI could probably done on the client side (browser) with some JavaScript. There is space on the file system for putting some js (or just enhancing style.css to make the update.html nicer), so that would fine, and it would not occupy code space on the TTGO.

LukePrior, 2021-08-08:
@dl9rdz I have updated but I don't see the update text on the about page.

[image]

The /update.html page does now show the version numbers in iframes, it would be good to make that request on the TTGO device and show the number without needing the iframes.

[image]

dl9rdz, 2021-08-08:
> @dl9rdz I have updated but I don't see the update text on the about page.

Yes, that is what the comment regarding the letters and numbers is about. The update only updates the software, not the files in the file system. The about page is a html file on the file system. You see that change only after a full re-flash or manually updating the file  (Because of that, the previous version is -B12, and the new one is -B13.)

LukePrior, 2021-08-08:
> Yes, that is what the comment regarding the letters and numbers is about. The update only updates the software, not the files in the file system. The about page is a html file on the file system. [...]

Oh yes, sorry all good now.

dl9rdz, 2021-09-16:
Looks nicer now (still client side, but I guess thats just fine)

closed 2021-09-16 by dl9rdz