dl9rdz/rdz_ttgo_sonde #125: Heltec V1 board: nothing on screen, no AP comes up

whallmann, 2021-08-08:
Hm, everything looks good if played with other software like peterus/LoraDigi on this board.
But uploading the rdz nov release and actual bins so nothing happens after uploading succesfull with esptool as instructed.
Also no wifi AP comes up. If i look with serial the upper left button requests an file upload by OTA. 
Thats all i see.
What is it, i made wrong?
73 Wolfgang DF7PN

whallmann, 2021-08-08:
**Remarks: the upload protocoll**

C:\Users\Wolfgang\Downloads\lora_TrackerD Update\Flashing_Firmware>esptool --chip esp32 --port COM5 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 RDZupdate.ino.bin
esptool.py v2.6
Serial port COM5
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: f0:08:d1:dc:6e:78
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 8MB
Flash params set to 0x023f
Compressed 1068288 bytes to 608467...
Wrote 1068288 bytes (608467 compressed) at 0x00001000 in 9.9 seconds (effective 861.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

whallmann, 2021-08-08:
accedently closed sorry.

dl9rdz, 2021-08-08:
> C:\Users\Wolfgang\Downloads\lora_TrackerD Update\Flashing_Firmware>esptool --chip esp32 --port COM5 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x1000 RDZupdate.ino.bin

That looks wrong. You can't use the update.ino.bin this way. You need to flash the full image (e.g. devel20210807-B13-full.bin) this way.

(If you want to flash the update using esptool, you have to make sure that previously you flashed (some previous version of) the full image, and then you have to flash the update at 0x10000 instead of 0x1000. )

whallmann, 2021-08-08:
Ups. Beginner failiure.
Will check this...
thanks
Wolf

whallmann, 2021-08-08:
Loading full image works . sorry

closed 2021-08-08 by whallmann