dl9rdz/rdz_ttgo_sonde #130: config SondeHub Settings

smarti2019, 2021-08-10:
Please make separately adjustable in the config chase car or sondehub transmission to sondehub. I would not like to constantly transmit qth via gps. Only decoded Data for Sondehub is my favorite.

dl9rdz, 2021-08-10:
Yes, thanks for the report, I noticed this recently myself. GPS position should only be sent if chase mode is active. Currently GPS position is sent always if no fixed position is entered.  Will be fixed in the next update.

closed 2021-08-10 by dl9rdz

LukePrior, 2021-08-10:
@dl9rdz can this be moved to the configuration file?

#define MIN_LOC_AUTO_DIST 200

It would be good to let the user set this range

dl9rdz, 2021-08-10:
The latest commit should solve this issue. Default value for sondehub.chase is currently 0, in which case no location information is sent. If you want to show up on sondehub, you have to set this value to 1 (fixed position), 2 (chase mode) or 3 (automated mode). Preferred mode is "auto", which operates in fixed position mode if no GPS position is available or if GPS position is close (200m) to your fixed position, and chase mode otherweise. 

Probably, we can change the default value for chose to 3 (auto) in the future. In any case, you have to manually have to enable sondehub upload, and in this case you can decide whether to use the default auto mode or set it to a different value.

Comments welcome.  Just lightly tested, so bugs are possible.

Maybe the distance used for switching from fixed to chase mode (in AUTO mode) should be in the config? Let me know what you think. I don't want to make the config excessively and unnecessarily complex, and 200m should be larger than accidental GPS accuracy erros, so it might be fine for everyone, but if a configurable value is considered useful, let me know.

dl9rdz, 2021-08-10:
@LukePrior ok, that was fast :-) Faster than I can write my note asking for comments. So I guess moving the distance to the config is on the TODO list.

(BTW... On holiday for about 2-3 weeks starting next week and busy finishing work stuff before the holiday, so don't worry if I disappear for a month or so with no updates or comments...)

LukePrior, 2021-08-10:
> @LukePrior ok, that was fast :-) Faster than I can write my note asking for comments. So I guess moving the distance to the config is on the TODO list.

Awesome, I've had a bit of a longer look through and I have a few more suggestions:

We don't need to send the uploader_position field if we don't actually have any data so this field can be removed: https://github.com/dl9rdz/rdz_ttgo_sonde/blob/devel/RX_FSK/RX_FSK.ino#L3144

This is unrelated to this PR but just a general comment I remembered when looking through, we should check if `uploader_contact_email` and `uploader_antenna` are set before uploading them.

You have added the position check code to station_update but not to the regular data upload in `sondehub_send_data`, this code should be changed to use the new checks you created for the station position upload: https://github.com/dl9rdz/rdz_ttgo_sonde/blob/devel/RX_FSK/RX_FSK.ino#L3325

I think that is everything.

> (BTW... On holiday for about 2-3 weeks starting next week and busy finishing work stuff before the holiday, so don't worry if I disappear for a month or so with no updates or comments...)

Have a nice time!

LukePrior, 2021-09-07:
Hi @dl9rdz I have updated my previous comment with all the suggestions if you get any time, I think we are getting close to releasing a new master version?

LukePrior, 2021-09-13:
These changes have all been made except for:

Move #define MIN_LOC_AUTO_DIST 200 to the configuration file with a note that it is in m

commit aeb3b1f3 (2021-08-10): automated fixed/chase mode for sondehub. also fix #130