dl9rdz/rdz_ttgo_sonde #131: LiveMap

14ri004, 2021-08-11:
small error in the LiveMap, I have lines that appear in all directions on the map, see photos, the function is fine, but some bugs, thank you

[image]
[image]

dl9rdz, 2021-08-11:
I guess this is just data with bit errors (invalid CRC). Next version should filter that out better.

14ri004, 2021-08-11:
Guten Abend, ich habe gerade den TTGO mit der neuen Version aktualisiert, teste heute Abend LiveMap und gebe dir morgen die Antwort, danke

good evening, I have just updated the TTGO with the new version, I am testing LiveMap this evening, and give you the answer tomorrow, thank you

14ri004, 2021-08-12:
hello, i tried the latest devel20210811-B14-full, and now it works perfectly, great job, thanks