dl9rdz/rdz_ttgo_sonde #132: Scan numbers are irregular when maxsonde is 11

TimMcMahon, 2021-08-12:
**Describe the bug**
Scanner cycles between two active QRG entries with ID 1 and ID 2 but it displays Scan 13 and Scan 2 respectively when maxsonde is set to 11. 

**To Reproduce**
Steps to reproduce the behavior:
- Add active QRG entry ID 1 with 401.5 and RS41. 
- Add active QRG entry ID 2 with 402.5 and RS41. 
- Deactivate other 18 QRG entries. 
- Change Maxsonde from 20 to 11 in Config menu and Save
- View Scanner on device
- Scan 13: RS41 appears and then
- Scan 2: RS41 appears

**Expected behavior**
I expected to see Scan 1: RS41 and then Scan 2: RS41 on the Scanner. 

**Version and specific configuration:**
 - devel20210811
 - Screen config: 1
 - QRG screen has active entries in ID 1 and ID  2
 - Maxsonde changed from 20 to 11
 - Fingerprint 31 (TTGO LORA32 v2.1_1.6 (0.9" OLED@21,22))

dl9rdz, 2021-08-12:
Ok, good spot :), thanks for the report.
If you do a reset after changing maxsonde, everything should work as expected. 
In any case, I will put it on the TODO list for fixing.

14ri004, 2021-08-12:
hello, I also have a problem, above 11 frequencies in the scan it does not take, on livemap it is identical up to 10 no worries, but not taken into account above 10 frequencies register

and second livemap and color problems on my phone, but black and white on my PC ??

[image]

F5MVO, 2021-08-12:
a french station using 20210809 devel decoding M20 when is M10 and blocking the decoding on sondehub.

dl9rdz, 2021-08-12:
> a french station using 20210809 devel decoding M20 when is M10 and blocking the decoding on sondehub.

Can you provide more details?
Do you know for which sonde this was happening? Why is it "blocking the decoding"?

F5MVO, 2021-08-12:
20210809 devel 

[image]

F5MVO, 2021-08-12:
I have to do a CTRL F5 to unlock the local decoding on sondehub

dl9rdz, 2021-08-12:
Ok I see. So the issue is that in either M10 or M20 setting, the decoder will decode any Meteomodem sonde, but report the setting type to sondehub. I have fixed that in the latest version, so the real decoded type will be reported to sondehub. (I hope... No testing has been done yet).

No idea why slightly wrong data could cause any larger form of issues (lockup) on sondehub though, maybe @LukePrior knows more about that. In any case, after a software update of the affected station, the problem should be gone.

F5MVO, 2021-08-13:
Hi, it seems ok with the last version 20210811, for M10 to Sondehub now, thanks

F5MVO, 2021-08-13:
Its seems ok with 20210812 also.

F4IRT, 2021-08-18:
#

> Ok I see. So the issue is that in either M10 or M20 setting, the decoder will decode any Meteomodem sonde, but report the setting type to sondehub. I have fixed that in the latest version, so the real decoded type will be reported to sondehub. (I hope... No testing has been done yet).
> 

Hello,
If the decoder sends the actually decoded Meteomodem type of probe (M10 or M20) to Sondehub, then the choice of M10 or M20 mode in QRG can be replaced by a single choice "M10 / M20". Cordially.