dl9rdz/rdz_ttgo_sonde #135: Current devel build fails on VSCode

DL2MF, 2021-08-14:
Trying to compile and make the current build fails with:

libraries\SondeLib\Display.cpp: In member function 'virtual void ILI9225Display::begin()':
libraries\SondeLib\Display.cpp:355:13: error: expected type-specifier before 'Arduino_ESP32SPI'
   bus = new Arduino_ESP32SPI( sonde.config.tft_rs, sonde.config.tft_cs,
             ^
libraries\SondeLib\Display.cpp:358:13: error: expected type-specifier before 'Arduino_ESP32SPI'
   bus = new Arduino_ESP32SPI( sonde.config.tft_rs, sonde.config.tft_cs,
             ^
*** [.pio\build\ttgo-lora32\lib812\SondeLib\Display.cpp.o] Error 1

dl9rdz, 2021-08-14:
I am not an expert con VSCode, but I assume you are using the PlatformIO plugin for VSCode.

Checking out a fresh copy of the devel branch and compiling it with PlatformIO on the command line with "pio run" compiles fine, so its likely not a problem with the source code.  Also the TravisCI build (using Arduino IDE platform) ran fine.

I vaguely remember having seen that you need to use the build button on the bottom in the VSCode platformio tool bar, not the VCcode build button on the top of the window, otherwise you get strange errors. But no idea if that possibly relates to your problem?

dl9rdz, 2021-08-14:
The error seems to indicate that the Arduino_ESP32SPI type is not defined. This type should be part of the Arduino_GFX library that has been added to platformio.ini as dependency back in April. Arduino_GFX_Library.h is included from Display.h which is included from Display.cpp, so all *should* be fine. Maybe if you used an older version before, you have to somehow tell VSCode to refresh the dependencies.

DL2MF, 2021-08-14:
Hi Hansi, 
thanks for your support and rapid reply. My VSCode/platform.io has been first updated today, before trying to make a build.
Repeating clean and build several times does not the trick like in some other projects where im also using VSCode.

The Arduino_ESP32SPI seems also to be in place and defined:
Compiling .pio\build\ttgo-lora32\lib713\GFX Library for Arduino\databus\Arduino_ESP32SPI.cpp.o
Compiling .pio\build\ttgo-lora32\lib713\GFX Library for Arduino\databus\Arduino_ESP8266SPI.cpp.o
Compiling .pio\build\ttgo-lora32\lib713\GFX Library for Arduino\databus\Arduino_HWSPI.cpp.o

I checked the latest changes in your GitHub already this morning and my guess was, the display.h is missing a definition, but looks like you already checked. The latest version I used was from 2021/04/11 and it did compile and build fine.

I'll look for an installation on another machine running linkux instead of windows to check and fix it.

73 Mike, DL2MF

DL2MF, 2021-08-17:
Made a new build on an older installation of VScode with fresh install of Arduino and platformIO IDE, everything works fine and compile finished successfully.

closed 2021-08-17 by DL2MF