dl9rdz/rdz_ttgo_sonde #136: Mavlink output for use with QGroundStation

gibbi, 2021-08-14:
It would be nice to have Mavlink output, either UDP and/or Serial (https://mavlink.io/en/). This enables us to use tools like QGroundstation to track Sondes easily. Especially the possibility to download maps in advance is very helpfull when tracking a Sonde far from the next celltower.

gibbi, 2021-08-14:
i already have a working udp implementation - its just breaking the display for some reason.
[image]

gibbi, 2021-08-14:
#137

eben80, 2021-08-31:
Hi! I'm curious what you want to do with this? Chase a falling sonde with a drone? That sounds awesome :D

gibbi, 2021-08-31:
main benefit is the great support for offline maps when chasing. more than once it felt like i was chasing cellular signal instead of the sonde. If i have someone chasing with me they just have to install the app or install it on a laptop and connect to the wifi and they can help me easily. 

Also at my workplace i can just feed mavlink into a very big PTZ Camera to get some pics. Yeah there are plenty of fun things you can do with this :-)

eben80, 2021-08-31:
Aah now I get it. Yes I normally also have to rely on offline maps on my mobile and a hotspot to which the ttgo is connected, especially in the mountains. I was just thinking you would be feeding the coordinates with mavlink into a drone autopilot and would be doing some fpv chasing as it falls :D

dl9rdz, 2021-08-31:
I will have a look at it when back from holiday. It certainly should be feasible to add it as optional code. I will not 1:1 accept the pull requests as it as default disable some of the features that should be in the standard binary image.

Depending on the number of users for this interface, it could either be optional (so you have to enable it in the source code and compile yourself) or if space permits part of the standard image.

As for offline maps, I have the impression that most people use Locus Maps, which work fine with the live klm feed from the TTGO.  the OSM app works as well, but only with manual position updates.

gibbi, 2021-08-31:
yeah - its still wip. i hope to have some good winds some day soon so that i can test this feature properly in the field and integrate it in the configuration. there is no need for a propper review before i do.

darksidelemm, 2021-09-17:
Also just putting out there that Chasemapper exists: https://github.com/projecthorus/chasemapper
(Also does offline maps, and does offline forward balloon flight path predictions)