dl9rdz/rdz_ttgo_sonde #138: sonde RS92

14ri004, 2021-08-17:
hi, on the Livemap the RS92 probe is not listed and is not visible, so what is in the list of the TTGO scan,

[image]

dl9rdz, 2021-08-17:
Thanks for the observation.

That is a combination of two artefacts: The RS92 decoder always sets validID to true, even if no frame was decoded (lat and lon is 0 in that case), and in that case the live map does not update the page content (if validID is true but lat or lon is 0). As soon as an RS92 is decoded, the display on the live map should be correct.

14ri004, 2021-08-17:
Thanks for the reply, it amazed me not to see the RS92 show in the livemap scan, like the M10 / RS41, okay i will test asap if it shows up, thanks

closed 2021-08-18 by 14ri004