dl9rdz/rdz_ttgo_sonde #140: decoded probe

14ri004, 2021-08-24:
hello, when I decode 1 single probe at a time, Example on 404.000, and I want to change the frequency to decode another probe on 405.700, the TTGO continues to decode the first probe and does not switch to the new frequency, this same if I uncheck the 404.000 box and I activate the 405.700, I have to reboot the TTGO so that it does the decoding, and this each time I change the frequency, it keeps in memory the first decoded probe, there is is there a solution or not to this problem ??, thank you

dl9rdz, 2021-08-24:
Not sure if I understand exactly what you are doing.
Did you just change the list of frequencies with the check boxes of active sondes? This list is used in scan mode, so if you change that list while a sonde is being decoded, nothing will happen. If that is what you are observing, it is as expected.

However, if you either switch back to scan mode (double-click on button 1) or simply switch to the next frequency (single press of button 1), this should activate the new frequency. You can do that using the web interface (Control tab) as well.

14ri004, 2021-08-24:
good evening and thank you for the answer, indeed that's exactly what it does, so I'm going to try using the control buttons, I'm using a TTGO 2.16 without buttons below, but with the software it should be good, thank you

closed 2021-08-24 by 14ri004