dl9rdz/rdz_ttgo_sonde #142: livemap

14ri004, 2021-08-31:
**Describe the bug**
hello, yesterday I had a little problem with the livemap, see photos

version devel20210817

[image]
[image]

LukePrior, 2021-08-31:
I have a feeling that this has to do with latitude being provided incorrectly, I haven't checked the code but if it's not in the range -90 to 90 you may see things like that.

14ri004, 2021-08-31:
hello, sorry for yesterday, I took the wrong forum, this is the first time he has indicated this to me, he goes around the world to mark the landing point just next to the probe, very strange as an indication , a little bug probably, thank you

LukePrior, 2021-09-05:
Can you test this again on the latest version?

closed 2021-09-14 by 14ri004