dl9rdz/rdz_ttgo_sonde #143: OLED 1.3" misalignment

tk5ep, 2021-09-01:
Hi all,
I'm new here and have assembled a T-BEAM v1.1 and 1.3 OLED display.

I've used the latest binary to program the ESP32.

The display is misaligned, it is shifted a bit to the left. A bit is missing at left and i have a weird vertical line at right. 
It seems to be OK verticaly.

I tried with an 0.96" display and it worked perfectly.

Is this related to the display or can it be corrected by soft somehere ?

Thanks,
Patrick TK5EP

dl9rdz, 2021-09-01:
This sounds like you are using a SH1106 display with the SSD1306 configuration.  The autoconfiguration always assumes you have a SSD1306 display (if a OLED display is detected), so if you use a SH1106, you have to manually adjust the configuration. Change "display type" from 0 to 2.

https://github.com/dl9rdz/rdz_ttgo_sonde/wiki/Hardware-configuration

tk5ep, 2021-09-02:
Thanks Hansi,

That worked perfectly ! Looks simple if you know it. :-)

Now i have to deal with the poor global sensitivity. Even with the balloon on sight, the receiver is losing the signal.

73 Patrick,

tk5ep, 2021-09-02:
Thanks for the help !

ToSStudio, 2024-08-21:
... same problem here.
Putting "2" in Screen Config doesn't change anything.
Can't find display-type in Config ....

hr-ru, 2024-08-21:
"Display type" is now in "Hardware configuration"

ToSStudio, 2024-08-22:
Tnx.
Works now.
Change should be reflected on Wettersonde.net tutorial, too