dl9rdz/rdz_ttgo_sonde #149: Website tabs overlapping on small screens

LukePrior, 2021-09-02:
**Describe the bug**
The page tabs on the web interface overlap when using a narrow screen such as a mobile device.

[image]

**Expected behavior**
I think it would be good that if the screen is too narrow two rows are shown. This can be observed when resizing the page on desktop quickly.

**Version and specific configuration:**
 devel20210817

dl9rdz, 2021-09-02:
Yes I noticed that on my smartphone as well. I am open for design suggestions.

Two lines might be an option, but would use up much space on a small smartphone display so it is not my favorite option.

As we now have the integrated lLiveMap, maybe the "Map" could be removed (there are direct link to several external mapping services in the Data tab anyway), and LiveMap renamed to Map?

Maybe the font size of the tabs could be adjusted depending on the screen width?

LukePrior, 2021-09-02:
> Two lines might be an option, but would use up much space on a small smartphone display so it is not my favorite option.

Yeah good point

> As we now have the integrated lLiveMap, maybe the "Map" could be removed (there are direct link to several external mapping services in the Data tab anyway), and LiveMap renamed to Map?

That sounds ok the only reason I added it was due to the lack of a local map, just be sure to remove all the code

> Maybe the font size of the tabs could be adjusted depending on the screen width?

This should probably also be added

eben80, 2021-09-08:
Hi all,
I'm no UX designer by any means but I came up with some kind of responsive menu.
When there is enough space to display the menus in landscape orientation, it will display the full menus:
[image]
When the device is in portrait orientation and there is not enough space to display the menus, it is collapsed into a "hamburger menu"
[image]
[image]

If it is of interest to you I could do a pull request with the changes. Of course it can be even more customized.

eben80, 2021-09-09:
Changes [HERE](https://github.com/eben80/rdz_ttgo_sonde/tree/responsiveui)

LukePrior, 2021-09-14:
I like the look of it much better than the existing solution.

eben80, 2021-09-14:
OK I'll create a PR for it and you can configure it as you like. Just remember to see changes a Shift+F5 reload is needed unless you change the stylesheet name.

eben80, 2021-09-16:
@LukePrior did you try it out already?

LukePrior, 2021-09-16:
> @LukePrior did you try it out already?

I updated with OTA and did a Shift-F5 refresh but didn't see the changes so I'm assuming I will need to manually flash it.

eben80, 2021-09-16:
Oh yes I see the OTA is from the 14th so perhaps the new changes are not included yet. Manual flashing should have it now.

dl9rdz, 2021-09-16:
The OTA update does not update the files (stuff in RX_FSK/data), only the code part.
If you don't want to re-flash, you could also do the update for the relevant file (style.css) via WiFi with 
```
cd RX_FSK/data
python3 ../../scripts/ttgoconfig put file style.css
```
add `--ttgo=<ip>` if you have a system without working mdns discovery

LukePrior, 2021-09-16:
Ok,

I have updated and can see the changes. I do note that there seems to be a small issue of the elements not lining up:
[image]
[image]
Not sure if these are intentional.

eben80, 2021-09-17:
Those are not intentional, I think they came it when I included the hamburger icon in css. Will see if I can match it exactly like before soon.

LukePrior, 2021-09-17:
> Will see if I can match it exactly like before soon.

Just tested and works perfectly Thanks

closed 2021-09-17 by LukePrior