dl9rdz/rdz_ttgo_sonde #154: devel20210908b

14ri004, 2021-09-10:
Hi, I updated my ttgo with the devel20210908b, but I have a crash and the functions are blocked, force revert to the devel20210908 version

eben80, 2021-09-10:
Did you flash via OTA? If not did you also update the filesystem?
I had a successful flash via VSCode upload for the 2.1.6 TTGO.

14ri004, 2021-09-10:
Hello, ttgo 2.16 I did the update by OTA, I tried with the devel, and redo a complete installation with the update, same problem, I redid the installation with the version devel20210908, and that works perfectly, I don't understand what's going on

eben80, 2021-09-10:
Strange....Have you got anything connected to IO35?

14ri004, 2021-09-10:
No nothing plugged in, I just received it 3 days ago, I had updated with the last update, and it works perfectly, it's only when I updated version devel20210908b, that the problems have arrived, I have 2 other identical TTTGOs and I am not the problem ??, for the moment it is not very serious since it works, I will not update it for the 'moment, I will wait for a new version to see, thanks for your help

DB2FM, 2021-09-10:
I have the same problem.
It seems to work in the background as I can connect via browser but the display can’t be switched as you always see the same Scan-Entry on the display.
Seems not to recognize buttons (besides RESET).

Nothing connected on any IO-port (nothing additionally ;) )

eben80, 2021-09-10:
Yes I can confirm the same behaviour using the OTA method or flashing the full bin files from here with the ESP32 Download tool.
When I compile the devel branch from here and upload it manually, it works fine.
 I do however always build with `board_build.partitions = no_ota.csv`

dl9rdz, 2021-09-10:
Thanks for the information. Yes, I can reproduce this with the binary image. If I compile the same code locally either with platformio or with ArduinoIDE, it works fine. So this will require some more investigation... Meanwhile, i have removed the 20210908b from the downloads page. The code in the repository should nevertheless be fine, the (b) was just a minor change for adding the possibility of showing the battery voltage for the small TTGO v2.1_1.6 (or similar boards with a voltage divider on the battery to an adc input)

DB2FM, 2021-09-10:
Tried to install (OTA) devel20210910test-B14, which resulted in sonde not working at all (no display, no contact from the web-browser).
Downloaded latest rdz_ttgo_sonde-devel but get errors on compiling:

Compiling .pio/build/ttgo-lora32/lib215/GFX Library for Arduino/display/Arduino_HX8347C.cpp.o
In file included from .pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.cpp:5:0:
.pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.h: In constructor 'Arduino_HWSPI::Arduino_HWSPI(int8_t, int8_t, int8_t, int8_t, int8_t, SPIClass*, bool)':
.pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.h:60:23: error: 'Arduino_HWSPI::_miso' will be initialized after [-Werror=reorder]
   int8_t _sck, _mosi, _miso;
                       ^
.pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.h:56:13: error:   'SPIClass* Arduino_HWSPI::_spi' [-Werror=reorder]
   SPIClass *_spi;
             ^
.pio/libdeps/ttgo-lora32/GFX Library for Arduino/src/databus/Arduino_HWSPI.cpp:27:1: error:   when initialized here [-Werror=reorder]
 Arduino_HWSPI::Arduino_HWSPI(int8_t dc, int8_t cs /* = -1 */, int8_t sck /* = -1 */, int8_t mosi /* = -1 */, int8_t miso /* = -1 */, SPIClass *spi, bool is_shared_interface /* = true */)
 ^
Compiling .pio/build/ttgo-lora32/lib215/GFX Library for Arduino/display/Arduino_HX8347D.cpp.o
Compiling .pio/build/ttgo-lora32/lib215/GFX Library for Arduino/display/Arduino_HX8352C.cpp.o
Compiling .pio/build/ttgo-lora32/lib215/GFX Library for Arduino/display/Arduino_HX8357A.cpp.o
cc1plus: some warnings being treated as errors
*** [.pio/build/ttgo-lora32/lib215/GFX Library for Arduino/databus/Arduino_HWSPI.cpp.o] Error 1
====================================================== [FAILED] Took 15.16 seconds ======================================================

dl9rdz, 2021-09-10:
Thanks for testing. The test image worked for me (but I did not test OTA; only full image), but I am not happy with it (it uses arduino esp32 2.0.0 instead of 1.0.6), as it needs much more RAM. Free RAM reduced to about 60k instead of 130k. So I will downgrade to 1.0.6 for now, until I have more time to investigate the memory use increase.

For building with Arduino IDE, using esp32 2.0.0 requires updating the library ESPAsyncWebServer in Arduino library folder, otherwise you get an compile error (but I suggest for not to not use 2.0.0 anyway)

For building with PlatformIO, the last commits should have no effect at all, because the only thing that changed is the .travis.yml that controls the TravisCI build for the web site bin image using ArduinoIDE. Your error is due to an update (and bug in the code) of the Arduino GFX library. I will update platformio.ini to directly point to an older version for now.

LukePrior, 2021-09-13:
Hi,

I'm running into the same issue as @DB2FM when trying to compile with PlatformIO what change needs to be made?

dl9rdz, 2021-09-13:
Downgrade the GFX library to an older, working version.

in platformio.ini
    https://github.com/moononournation/Arduino_GFX#v1.1.5

Sorry, changed that locally on Friday and forget to push to github. Should be there now.

DB2FM, 2021-09-13:
Now it's working again :)
Thank you very much

eben80, 2021-09-13:
So it would be OK if I do the normal ESP32 platform upgrades in VSCode/Platformio too because you specify the version that you need right?

closed 2021-09-14 by 14ri004