dl9rdz/rdz_ttgo_sonde #155: Feature Request: Add signal level and timestamp of the latest received frame to the LiveMap

dm5tt, 2021-09-11:
Hello!

It would be great to see following informations on the LiveMap:

- Timestamp of the latest received frame
- Signal level (dBm?) of the latest frame

Thank you!
Holger / DM5TT

eben80, 2021-09-14:
Something like this?
[image]
The problem I have with it is that the rssi is not tied to the received frame, but a realtime value.  This means that  the livemap thinks new frames are received because the rssi value changes, even if that is not the case, so the status indicators keep green when they should be yellow.
Maybe the experts know how to fix it? :)

dm5tt, 2021-09-15:
Yep! That looks great!

But I can't help regarding the web interface :-( My knowledge is limited to embedded stuff.

eben80, 2021-09-16:
I think @cixio brought us the livemap functionality.

dl9rdz, 2021-09-18:
I had a look at the livemap implementation (which is not from me). It shows green dot if something changed, otherwise a yellow dot. So this is of course not suitable if RSSI is included (which is updated even if no new frame was decoded).

Probably the best option would be to include the decoder status. This is used by the Android app, and has three states (timeout=>nothing received; crc error=>valid sonde header detected, but not successfully decoded do to bit errors; ok).

A simpler alternative is to just check if lat/lon has changed. Even if the sonde is laying on the ground this should have some noise and thus change between frames. Or even more easily include and check the virtual frame counter (sondeInfo->vframe. Even for sonde types that do not have a frame counter, this is a value that changes on every frame that is received.

dl9rdz, 2021-09-19:
Latest version sends res and vframe in live.json.
Also modified livemap.js such that it checks if data.res==0 instead of JSON.stringify(data) !=  JSON.stringify(last_data).

eben80, 2021-09-20:
The live map wasn't displaying a received sonde for me this morning. No green dots. Is it just me? 😬

dl9rdz, 2021-09-20:
Did you update to the latest latest version? 
(As I did one modification without increasing the version number, this means devel20210919 from after 20:30Z yesterday, not the devel20210919 that was there for two hours before, which had the "res" field in the wrong place, and thus could have the effect you describe)

eben80, 2021-09-20:
Yes I think I missed your last commit. So I'll check it out with the latest
changes thank you.

eben80, 2021-09-20:
I've created this PR for this and tested the functionality.
https://github.com/dl9rdz/rdz_ttgo_sonde/pull/182
I needed to change the structure of live.json to be able to use the rssi value.
The sonde detail will remain even if the signal is lost and I also added last heading.
[image]
.

eben80, 2021-09-21:
There's just one thing missing for me. Previously, if I left the ttgo on overnight and opened the live map from fresh in the morning, it would show the last position. Now it won't, because its only using the res value and not checking if there's any existing data. It would be nice if it loaded any data that is there when you first open the browser. For that to happen now, it would need to be actively receiving.

dl9rdz, 2021-09-21:
Ah thanks. II hope with the latest commit I have restored the previous behaviour in this regard.

eben80, 2021-09-21:
> Ah thanks. II hope with the latest commit I have restored the previous behaviour in this regard.

I meant missing on my own PR as well of course 😁

LukePrior, 2021-09-21:
The map doesn't seem to show the current launch site and frequency when no sonde is being decoded at the top as it was previously.

[image]

eben80, 2021-09-21:
> The map doesn't seem to show the current launch site and frequency when no sonde is being decoded at the top as it was previously.
> 
> [image]

It's this:
`if (data.lat == '0.000000') { return false; }`

I'm changing my PR to fix the display of historical data at first browser launch, if it exists.

dl9rdz, 2021-09-21:
Yes. I just removed it. But now lets wait for the PR. :)

eben80, 2021-09-21:
:D OK just going to get an old sonde from my car to test.

eben80, 2021-09-21:
OK I hope that's sufficient.
I tried to test for all scenarios I can think of and the only time you might get an anomaly is with rssi value if you have a cold start of the browser and previous data exists from live.json. It will give you the current rssi.
I think that's not a big deal though.

LukePrior, 2021-09-21:
Seems good to me now.

eben80, 2021-09-21:
@dm5tt do you have a chance to check? Can this one be closed?

dm5tt, 2021-09-22:
Will check in 1.5 weeks, as In travelling at the Moment. Thanks all for the work!

dm5tt, 2021-10-03:
Back! Looks great! Thank you guys/girls!

The added informations are really useful to improve the setup.

closed 2021-10-03 by dm5tt