dl9rdz/rdz_ttgo_sonde #159: devel update

14ri004, 2021-09-13:
a display bug, after downloading the new devel

[image]

[image]

LukePrior, 2021-09-14:
Hey,

Can I see a copy of your configuration settings for SondeHub as that was where all the updates were made.

Cheers

14ri004, 2021-09-14:
hello, it's a ttgo t beam

[image]

LukePrior, 2021-09-14:
Hmm it must be related the code reformat and not the SondeHub changes as you have that disabled.

dl9rdz, 2021-09-14:
This is not a new problem, this has been that way all the time.
The OTA code is from a time where only OLED displays were supported, and the y position (lines on the OLED) is hard-coded in +1 increments (for the TFT display, it is now "pixels").

14ri004, 2021-09-14:
Thank you for your answers, it was just to inform you of the problem on the screen, it does not bother the operation of the TTGO

closed 2021-09-14 by 14ri004