dl9rdz/rdz_ttgo_sonde #162: Roadmap for master 0.9 and master 1.0

dl9rdz, 2021-09-14:
Plan is to have a final master 1.0 by end of 2021, and prior to that a mostly feature-complete master 0.9 around October.

Open issues that possibly (likely?) will be added until then:

Decoders
- [ ] Check RS41 data quality (clear subframe buffer if a new sonde is received)
- [ ] Check DFM data quality (clear all data and wait for data completion if a new sonde starts being received)
- [ ] Possibly enhance humidity meteo data for RS41 (no priority for me, will accept contributions) #152
- [ ] Show RINEX status on web interface #65
- [ ] Maybe replace M10 and M20 by a single M10/M20 entry in frequency config (both use exactly the same decoder)

Board and display support
- [ ] Add support for M5STACK Core Gray, test and enhance version for M5STACK Core Due
- [ ] Add compass display for OLED display (merge existing code from somewhere...)
- [ ] Fine-tune screen2 for landscape TFT
- [ ] Optimize screen 3 for portrait TFT (current version is very rudimentary, contributions welcome!)
- [ ] Add better screen 4/5 of for large ILI9341 TFT (hopefully will get a contribution for that)
- [ ] Add new screen 6/7 for small but high resolution ILI9342 TFT (M5STACK)
- [ ] Fix line spacing on TFT during OTA update

Network interfaces
- [ ] Test MQTT interface and close open issue #58
- [ ] Enable aprs/radiosondy upload of data
- [ ] Finish SondeHub interface
- [ ] Test if altitudemode in KML works fine also with LotusMap live feed
- [ ] Add features to integrated HTML map (show more info like time of last frame)
- [ ] Maybe redesign qrg.html with AJAX/JSON requests to make network message smaller (less out-of-memory issues)
- [ ] Mabye do the same for config.html (same reason)

Documentation
- [ ] Update documentation (and add documentation how to easily contribute to documentation) #88
- [ ] Add wiring diagrams to documentation

Others
- [ ] Offline maps for rdzwx-go
- [ ] Move SondeHub static gps position to global static gps position that is also used for compass on TFT if no GPS availabl
- [ ] Reorganize flash layout: More space for code, separate partition for TFT fonts (exclude them from OTA updates), smaller partition for file system. 
- [ ] User friendly flash tool (add features, create executable with pyinstaller for Mac/Windows)
- [ ] (Maybe) download of active scan frequencies from sondehub? #120 
- [ ] MAYBE: Check if the new arduino esp32 idf version can be used (seems to use much more RAM, so maybe not practical?) (not yet available for platform io yet anyway, so maybe for post-master-1.0)

LukePrior, 2022-01-02:
Can we get a new master release from the devel branch as 0.9 has several bugs in regards to SondeHub uploading.

TarmoKolli, 2022-01-08:
Maybe it would be possible to add SD card support and write the sonde location information to the card as well. I'm looking for landed sondes and would be very interested in the all latest GPS location points.

LukePrior, 2022-01-22:
> Maybe it would be possible to add SD card support and write the sonde location information to the card as well. I'm looking for landed sondes and would be very interested in the all latest GPS location points.

The best way to do this would be storing the MQTT messages on an external device or you could upload the data to SondeHub and use the APIs to download it.

TarmoKolli, 2022-02-08:
> The best way to do this would be storing the MQTT messages on an external device or you could upload the data to SondeHub and use the APIs to download it.

I send my received signals to both tracker.sondehub.org and radiosondy.info servers, but since I don't have a HAM license, I send data every 45 seconds. The servers are blocking me from sending data at shorter intervals.
Maintaining a separate data server, raspberry or pc would reduce the benefits of rdz_ttgo.

dl9rdz, 2024-09-04:
This has become somewhat outdated.

Ongoing work on an upgrade in the new dev branch with new esp32 idf will result in a version main1.0 later this year.  Maybe a new roadmap for the future will be planned based on that....

closed 2024-09-04 by dl9rdz