dl9rdz/rdz_ttgo_sonde #163: Sondehub Include RFC7231 Date Header & User-Agent

LukePrior, 2021-09-15:
We should include a UTC date header and User-Agent string when uploading data in `sondehub_send_header`. The UTC date header can be determined the same way as `datetime` is using `ts`. The header should be in the RFC7231 format which is `Date: <day-name>, <day> <month> <year> <hour>:<minute>:<second> GMT` for example `Date: Wed, 21 Oct 2015 07:28:00 GMT`. This header is used to help find uploaders with incorrect system times.

While not strictly neccessary it would also be good to send the User-Agent header with the format `SOFTWARENAME-SOFTWAREVERSION` for example `rdzTTGOsonde-devel20210914` or `autorx-1.4.1-beta5`. I'm not completely sure how to include these so if anyone knows how they can be added that would be great.

dl9rdz, 2021-09-16:
Should be included in the last commit.
RFC7231 asks for `SOFTWARENAME/SOFTWAREVERSION`, so I decided to use that.

LukePrior, 2021-09-16:
Perfect, I can confirm everything looks good on our end.

closed 2021-09-16 by LukePrior