dl9rdz/rdz_ttgo_sonde #167: Suggestion: Should packet checks be moved out of SondeHub to somewhere common?

LukePrior, 2021-09-17:
So we currently do a few checks on the data before we upload it to SondeHub.

I saw that you were hoping to fix up exporting for the 0.9/1.0 release and was wondering if it would be a good idea to move these checks to somewhere common so they can benefit all uploads?

dl9rdz, 2021-09-17:
Checking if lat/lon is 0 or alt > 50000: Sure, this makes always sense to filter out.

As for the others, it probably depends on the uploads. Maybe for "public" uploads (but there SondeHub is the only one currently, even if aprs.fi/radiosondy could easilly be enabeld). Not so much for "private" ones (Android app, aprsmap, private mqtt server, etc.)

Sat >=4. For private feeds, I'd like to see data if a sonde transmits some lat and lon values and sat<4 in its telemetry. For public feeds, maybe yes, as there is some chance of lat/lon being simply wrong.

Time filter. Even less sure. For any private feed or local display, I want to see valid lat/lon data even if for some reason the time stamp is corrupt. Or if the TTGO failed to obtain current time from a NTP server.

LukePrior, 2021-09-17:
> As for the others, it probably depends on the uploads. Maybe for "public" uploads (but there SondeHub is the only one currently, even if aprs.fi/radiosondy could easilly be enabeld). Not so much for "private" ones (Android app, aprsmap, private mqtt server, etc.)

Oh yeah, I was mainly thinking of aprs/radiosondy uploading which you indicated as 0.9/1.0 feature.

I feel like there could be value in moving the code out and all public uploads need to run through it but there is a user setting disabled by default which also routes local packets through?

I would imagine this would save space instead of implementing it twice for SondeHub and APRS while giving users more choice if they want it.