dl9rdz/rdz_ttgo_sonde #172: Western Hemisphere Longitude Issue with Burst and Landing Predictions

WH3080, 2021-09-18:
I am truly impressed how this software has evolved. I am using it in North America. For the first time the prediction function now works for me on the live map (not on the rdzwx-go app though). 
While the predicted landing site is accurately displayed on the live map, the indicated longitude is always off. That means forwarding to Google Maps or other maps doesn't work. For example when Lat is 50N, Lon should be something like 110W (i.e. -110), instead 248W is displayed, which is not plausible. I believe there is a calculation bug somewhere which is not apparent to Eastern Hemisphere users that mainly use your fantastic software.

dl9rdz, 2021-09-18:
Thanks for the report. Indeed, our previous fix only solved parts of the problem, the label and link to maps still contained the value as returned from tawhiri, instead of moving it to the -180..180 range.
Should be fixed in the next version. Make sure to refresh the browser cache.

WH3080, 2021-09-19:
Thank you! Your prompt help is greatly appreciated!

dl9rdz, 2021-09-22:
Closing this issue, as it should be fine now.

closed 2021-09-22 by dl9rdz