dl9rdz/rdz_ttgo_sonde #176: Decoding RS41 Subtype and Battery Voltage

LukePrior, 2021-09-19:
We already decode the subtype for DFM sondes so it might be a good idea to also decode it for RS41 sondes. I don't think we decode battery voltage for any sondes but I think it could be useful in determining if a sonde will stop transmitting soon. I have found some examples of other programs decoding these values.

**RS41 Subtype:**

- https://github.com/einergehtnochrein/ra-firmware/blob/master/src/rs41/rs41private.h#L305
- https://github.com/projecthorus/radiosonde_auto_rx/blob/master/demod/mod/rs41mod.c#L331

**RS41 Battery Volatage:**

- https://github.com/einergehtnochrein/ra-firmware/blob/master/src/rs41/rs41calibconfig.c#L141
- https://github.com/projecthorus/radiosonde_auto_rx/blob/master/demod/mod/rs41mod.c#L416
- https://github.com/oe5hpm/dxlAPRS/blob/master/src/sondemod.c#L3574

I have tried to link to examples of how this is implemented in other programs but I can't fully figure out how to implement it here so if you would like to try that would be great otherwise feel free to close this issue.

mycarda, 2021-09-19:
I have added some code for subtype and battery voltage.

Subtype is part of the subframes we already capture for the temperature and humidity calibration so that's easy enough. The battery voltage is included in the 79-status frame and extracting that is easy enough, too. I just need to test the code works and then I will create a pull request.

Testing for me will be at 6:30 and 10:30 tomorrow from the Larkhill launches.

As an aside, I create two new variables for subtype and battery voltage in sonde.h but I noticed there was already a variable called subtype in sonde.h but that is a int8_t so not useful for storing a char[10]. I called my new variable variant. There is also a variable called typestr but that's not large enough either. I thought it was safer to create two more variables rather break something else. Let me know if this is the wrong approach.

dl9rdz, 2021-09-19:
For the subtype, adding a char[10] to Sonde.h would work, but I am a bit worried about the memory footprint. For e.g. maxsonde=50, thats 500 bytes extra RAM, as it is per qrg slot, for information that is anyhow already stored in the subframe buffer.  

So instead, in the last push, I added my preferred version, which is a function to RS41 that returns the subtype directly from the subframe buffer, if the corresponding data is valid.

For the battery adding a field in sonde.h would fine, as I guess that could be used for other sonde types as well.

mycarda, 2021-09-19:
> So instead, in the last push, I added my preferred version, which is a function to RS41 that returns the subtype directly from the subframe buffer, if the corresponding data is valid.

Yes, that sounds like a much better way of doing it.

I have pulled your changes and I will add the battery voltage code on top and test tomorrow.

LukePrior, 2021-09-20:
> So instead, in the last push, I added my preferred version, which is a function to RS41 that returns the subtype directly from the subframe buffer, if the corresponding data is valid.

Hey this works pretty much perfectly except sometimes a subtype of `XDtS3agy-Y` is returned, any idea why that could be?

mycarda, 2021-09-20:
Any idea why the communications speed was changed on the serial port in RX_FSK.ino?

`Serial.begin(921600 /*115200*/);`

I thought I had broken something this morning. :-)

dl9rdz, 2021-09-20:
Ah sorry, this should not have made it into the public repo.
I used heap_caps_dump() to look at why arduino idf 2.0.0 uses much more RAM compared to idf 1.0.6, and this function is too slow with 115200 baud and causes a wdt. Will be back to 115200 in the next commit.

eben80, 2021-09-20:
> Hey this works pretty much perfectly except sometimes garbage subtypes such as `XDtS3agy-Y`, `6NeaEF287Z`, or `RS41-SGP7Z` (I've seen over 10 so far) are sent, any idea why that could be?

I've noticed this this morning too. Will create an issue for it.
[image]

mycarda, 2021-09-21:
Tested battery voltage reading on two launches this morning. Created pull request #183

dl9rdz, 2021-09-21:
Ok, merged (without updating version number). Did you also implement the transmission to sondehub (because that was not included in the pull request)?

mycarda, 2021-09-21:
> Did you also implement the transmission to sondehub (because that was not included in the pull request)?

No, I simply added and tested the voltage. I am not familiar with the sondehub transmission code.

eben80, 2021-09-21:
Is it as simple as adding this below?
```
    // Only send battery voltage if RS41 and not 0
  if ((realtype == STYPE_RS41) && ((float)s->batteryVoltage != 0)) {
    sprintf(w, "\"batt\": %.2f,", (float)s->batteryVoltage);
    w += strlen(w);
  }
```

dl9rdz, 2021-09-21:
Yes, basically what I just added.

LukePrior, 2021-09-21:
Those changes look good I will wait for some data to arrive in SondeHub and will close this issue. We will now be at feature parity with auto rx for RS41-SG sondes, just missing pressure.

LukePrior, 2021-09-21:
Everything seems to be working.

closed 2021-09-21 by LukePrior

LukePrior, 2021-09-22:
@eben80 Do you get RS41 launches with Ozone or other XDATA sensors in England?

eben80, 2021-09-22:
I'm in Slovakia and they are normally just standard RS41-SGP without any aux unfortunately.