dl9rdz/rdz_ttgo_sonde #178: Intermittent TTGO restarts for Stack smashing protection failure

eben80, 2021-09-19:
**Describe the bug**
This started I think with the recent humidity decoder update.  I've saved the serial output to log and it appears that the device runs into a buffer overflow when the temp and humidity values are calculated successfully and uploaded to SondeHub.

**To Reproduce**
When receiving an RS41-SGP sonde and uploading the data to SondeHub, everything seems to go fine until a valid humidity value is calculated and sent. The device restarts when the overflow occurs.  SondeHub import was also enabled but I don't currently see any correlation with that, looking at the logs.

**Expected behavior**
Device didn't reboot like this with the previous humidity decoder code.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version and specific configuration:**
 - I'm using devel20210917 with some minor changes ie. display extra data in the sondestatus web interface plus split up the conditional send of temp and humidity values to Sondehub
 - Fixed location, sondehub import enabled, sondehub upload enabled
 - Hardware you are using? (TTGO v2 1.6 with GPS module on pin 39, battery detection on pin 35 and two buttons.)

**Additional context**
After the second restart, the device restarted in AP mode. After this the device didn't restart again, even after having the humidity value available. The difference of course is that it didn't have an internet connection.

I've included the serial logs.
[putty.zip](https://github.com/dl9rdz/rdz_ttgo_sonde/files/7192641/putty.zip)

I know that there is some new changes pushed so I will flash the device with those changes and check if it still has this issue in the morning.

dl9rdz, 2021-09-19:
Ah ok great, thanks for the report, this is very useful and solves the mystery of reboots that I have being informed about recently.

The fix is simple. In RS_FSK change
#define MSG_SIZE 550
to something larger. 

(Or, as a temporary fix, remove your antenna description.  In the attached log, its just 3 bytes that are missing :) With the latest update that includes the RS41 subtype, its a bit more)

eben80, 2021-09-19:
Thank you! At least it seems like an easy fix :D 
Any suggestion on MSG_SIZE value? I can compile and flash it to test for the midnight flights.

dl9rdz, 2021-09-19:
I changed it to 1000. Should be fine for now.

eben80, 2021-09-19:
Great thanks. I'll update here in the morning.

eben80, 2021-09-20:
It works thank you.

closed 2021-09-20 by eben80