dl9rdz/rdz_ttgo_sonde #185: Strange Temperature & Humidity Values RS41

LukePrior, 2021-09-22:
I was looking at some data and saw something weird happening with a `devel20210921` station. The uploaded temperature and humidity values were sent as an incorrect constant for about a minute before matching auto_rx, could this be something with calibration values?

Here is the data:
[image]
[image]

Any ideas @mycarda @dl9rdz

mycarda, 2021-09-22:
@LukePrior
Is 02:53:45 the time we started sending the temperature and humidity values or are you just showing part of the whole chart. The reason I ask is I would expect the temperature to be available maybe 30 seconds before the humidity value.

Also, both the temperature and humidity start behaving nicely at exactly the same time. As humidity is dependent on temperature, if there is a problem with the calibration values or calculation it would point to the temperature being the issue although I would expect some variation in the temperature and humidity, not just a fixed value (which what appears on the chart).

Finally, when we decide to send the temperature
https://github.com/mycarda/rdz_ttgo_sonde/blob/8a62085f4f362a009a041abf6b52fc14756c2f0d/RX_FSK/RX_FSK.ino#L3720
how do we know the value is a valid calculation and not just anything other than zero?

Sorry. all questions so far and no answers.

mycarda, 2021-09-22:
> I would expect the temperature to be available maybe 30 seconds before the humidity value.

Having said that, the launch I monitored this morning happened to get all the humidity calibration blocks first and the temperature calibration ones last so it's not always the case - just usually. :-)

LukePrior, 2021-09-22:
Hey,

The serial number of the aforementioned sonde was: S3130783.
The station uploading the data was: Test_Heiko_1_1_RDZTTGO

> Is 02:53:45 the time we started sending the temperature and humidity values or are you just showing part of the whole chart.

The first packet from this station for this sonde was at 02:53:49 which is the first packet shown in the images above.

> I would expect some variation in the temperature and humidity, not just a fixed value

I can confirm that the value was constant which alerted me to the data initially.

- Humidity: 58.511
- Temperature: -42.99

> how do we know the value is a valid calculation and not just anything other than zero?

We do not, if you have any suggestions to improve that I would love to hear them.

> Sorry. all questions so far and no answers.

Not at all, thanks for your time looking into it.

eben80, 2021-09-22:
It happened with me yesterday lunchtime for the first few minutes with a devel20210920 build before it corrected itself.

The flight is attached below and the nice thing is you have data from 2 different auto_rx versions plus SondeMonitor(who has a problem with sonde bearing)
I didn't see the same problem with the last midnight flights.
[response_1632231373417.xlsx](https://github.com/dl9rdz/rdz_ttgo_sonde/files/7208612/response_1632231373417.xlsx)

LukePrior, 2021-09-22:
> SondeMonitor (who has a problem with sonde bearing)

This is undergoing updates and the latest is now 6.2.5.2 so that may well be fixed.

dl9rdz, 2021-09-22:
Ok, I will fix this (see #186 last item that I just added)

mycarda, 2021-09-22:
> We do not, if you have any suggestions to improve that I would love to hear them

Initially, I think the values are set to impossible values
https://github.com/mycarda/rdz_ttgo_sonde/blob/ab1da6ea9aacc4a507f31aeec816c9d043a8b590/RX_FSK/src/Sonde.h#L111-L113
So maybe check those.

I think it might be more reliable to explicitly mark the values as invalid when we check if we have the required calibration values
https://github.com/mycarda/rdz_ttgo_sonde/blob/ab1da6ea9aacc4a507f31aeec816c9d043a8b590/RX_FSK/src/RS41.cpp#L769-L773
by adding
`else si->temperature = -300.0` or maybe `si->temperature = NAN`

Similarly for humidity.

dl9rdz, 2021-09-22:
The wrong values are most likely old values from a previous sonde that remain there as long as we don't have new calibration data for calculating new values. 

I added some code that now clears all data, including temperature etc., whenever the calibration data is cleared (manual config change, new ID detected). This should solve this issue.

mycarda, 2021-09-22:
> Initially, I think the values are set to impossible values

I was wrong. The value of the temperature before we calculate it is 0.0 which is probably why the existing check works so well. Technically, this might be a valid value at some point in the flight as, technically, so is 0.0 humidity.

dl9rdz, 2021-09-22:
In the latest update, Sonde::clearAllData() sets all floats to NAN. that includes position and temp/humitidy.
Its called from RS41.cpp whenever a new ID is detected.
(I will have a closer look at the non-RS41 decoders, but it should be less an issue elsewhere)

LukePrior, 2021-09-22:
> In the latest update, Sonde::clearAllData() sets all floats to NAN. that includes position and temp/humitidy.

Will these checks need to be changed in that case?

https://github.com/dl9rdz/rdz_ttgo_sonde/blob/devel/RX_FSK/RX_FSK.ino#L3719

dl9rdz, 2021-09-22:
yes

closed 2021-09-22 by LukePrior