dl9rdz/rdz_ttgo_sonde #189: Missing info about receiving sonde in app

pavolgaj, 2021-09-24:
**Describe the bug**
Mobile App doesn't show any information about receiving sonde (in "i" window). It still displays frequency 400.0 MHz and sonde with id A... with red dot.
In the web interface, everything works and the data of the current sonde are there.

**To Reproduce**
Receive data from some sonde (I use RS41). Turn on mobile app and open info window.

**Expected behavior**
Show info about current sonde or cycle through all set frequencies and sondes.

**Version and specific configuration:**
I used the latest devel (devel20210924-B16). But, I also tried some older (devel20210922-B16). I didn't try all - there are a quite many. Finally, I came back to my previous version (devel20210916-B16) which works.

dl9rdz, 2021-09-24:
Thanks for the info. Yes, some recent reorganisation causes an issue there.  Should be "good enough" with devel20210924b" for now.

pavolgaj, 2021-09-24:
Now, it works fine. Thanks.

closed 2021-09-24 by dl9rdz