dl9rdz/rdz_ttgo_sonde #190: M10 Temperature Anomoly

LukePrior, 2021-09-24:
Hi,

I saw you added M10 Temperature decoding and I can see the data being uploaded to SondeHub however the values differ significantly from those of auto rx.

I'm not sure what the differences in code are but here is the function for M10 temp in auto rx: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/demod/mod/m10mod.c#L635

The value Anomoly was observed from sonde 010-2-12606 at frame 1316520110. The temperature reported by rdzTTGOsonde was -56 while the value reported by multiple auto rx stations was -62.5.

F5MVO, 2021-09-24:
i confirm, i run http://rdzsonde.mooo.com/devel/devel20210924b-B16-full.bin and lookat the temperature under my call F5MVO

F4CVN | ME0A22A2E | 2021-09-24 12:30:30z | 48.8225 φ | 2.13093 λ | 128 ° | 26 km/h | 23337 m | 5.3 m/s | -55.6 °C | -1.0 hPa | 0.0 % | mPa
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
F5MVO | ME0A22A2E | 2021-09-24 12:30:20z | 48.8227 φ | 2.13079 λ | 122 ° | 24 km/h | 23286 m | 6.0 m/s | -27.3 °C | hPa | % | mPa
M0HOW-10 | ME0A22A2E | 2021-09-24 12:30:00z | 48.8232 φ | 2.13001 λ | 130 ° | 4 km/h | 23164 m | 5.1 m/s | -55.3 °C | -1.0 hPa | 0.0 % | mPa
F5RZC-14 | ME0A22A2E | 2021-09-24 12:29:50z | 48.8235 φ | 2.12975 λ | 206 ° | 13 km/h | 23111 m | 5.5 m/s | -55.0 °C | hPa | % | mPa
F6ASP-11 | ME0A22A2E | 2021-09-24 12:29:30z | 48.8241 φ | 2.12976 λ | 197 ° | 24 km/h | 22990 m | 4.8 m/s | -55.1 °C | -1.0 hPa | 0.0 % | mPa
F5RZC-14 | ME0A22A2E | 2021-09-24 12:29:20z | 48.8244 φ | 2.12978 λ | 207 ° | 22 km/h | 22939 m | 4.8 m/s | -54.1 °C | hPa | % | mPa
14OA400 | ME0A22A2E | 2021-09-24 12:29:00z | 48.8249 φ | 2.13006 λ | 218 ° | 7 km/h | 22823 m | 6.1 m/s | -55.7 °C | -1.0 hPa | 0.0 % | mPa
F5MVO | ME0A22A2E | 2021-09-24 12:28:38z | 48.826 φ | 2.13 λ | 171 ° | 28 km/h | 22706 m | 6.0 m/s | -81.3 °C | hPa | % | mPa
14OA400 | ME0A22A2E | 2021-09-24 12:28:30z | 48.8266 φ | 2.12985 λ | 133 ° | 30 km/h | 22656 m | 6.9 m/s | -55.8 °C | -1.0 hPa | 0.0 % | mPa
JH66LU-11 | ME0A22A2E | 2021-09-24 12:28:00z | 48.8276 φ | 2.12977 λ | 262 ° | 17 km/h | 22486 m | 4.6 m/s | -58.3 °C | hPa | % | mPa
F5MVO | ME0A22A2E | 2021-09-24 12:27:58z | 48.8276 φ | 2.1297 λ | 254 ° | 19 km/h | 22491 m | 4.9 m/s | -37.7 °C | hPa | % | mPa
F4CVN | ME0A22A2E | 2021-09-24 12:27:30z | 48.8283 φ | 2.13119 λ | 228 ° | 15 km/h | 22344 m | 3.5 m/s | -59.1 °C | -1.0 hPa | 0.0 % | mPa

dl9rdz, 2021-09-24:
Ok, thanks for testing, I should never publish untested code.
(Indeed I carefully checked that the calculation is identical to autorx and dxlaprs before, but then a minor detail went wrong... I missed that the byte order for the temp adc value is different to the byte order of other telemetry data).
Should be fine in the latest commit and the re-built devel20210924b (I did not update the version number with the commit)

F5MVO, 2021-09-24:
its OK now

dl9rdz, 2021-09-24:
Seems ok, so closing this now.

closed 2021-09-24 by dl9rdz