dl9rdz/rdz_ttgo_sonde #192: RS41 use pressure sensor values in humidity calculation if available

mycarda, 2021-09-25:
I see code has been added obtain the pressure from the pressure sensor if the RS41 has one onboard. If the pressure from the sensor is available, this should be used to calculate the relative humidity rather than the estimated pressure.

https://github.com/dl9rdz/rdz_ttgo_sonde/blob/f30c9b6d89662f1aa5bd623eea7432a91d60596c/RX_FSK/src/RS41.cpp#L627

The estimated pressure should only be used if there is no pressure sensor onboard.

Maybe pass the pressure to the humidity calculation as an extra parameter then pass either the pressure from the pressure sensor (if available) or pass the estimated pressure value if there is no sensor onboard.

What I don't know is if the estimated pressure should be used until a valid pressure value is available and then use the "real" pressure in the calculation for relative humidity, or if it is better to wait until a valid pressure is available before calculating any relative humidity values rather than jump from estimated to real pressure values in the calculation.

LukePrior, 2021-09-26:
> or if it is better to wait until a valid pressure is available before calculating any relative humidity values rather than jump from estimated to real pressure values in the calculation.

I think this would be the best solution atleast in terms of uploading to SondeHub as we have so much data we only want to keep the perfect stuff.

dl9rdz, 2021-09-26:
Ok, should be that way now.

dl9rdz, 2021-10-05:
No complaints :) so I guess I can close this issue.

closed 2021-10-05 by dl9rdz