dl9rdz/rdz_ttgo_sonde #196: M10 type in qrg.txt indicated like M20 in DATA web server

F5MVO, 2021-09-28:
qrg.txt
402.004 M - Bordeaux 
401.000 M - Trappes         
402.000 M - Trappes         
404.800 2 - URY             
405.200 2 - URY    
==============================        

404.004 MHz, Type: M20
--
ID:
QTH: nan,nan h=nanm
Frame# 0, Sats=0, 1970-01-01 00:00:00
GEO-App - radiosondy.info - SondeHub Tracker - OSM - Google

F5MVO, 2021-09-28:
****

closed 2021-09-28 by F5MVO

dl9rdz, 2021-09-28:
As soon as a real signal is received, the real type (M10 or M20) will be shown. 

M10 and M20 are so similar that in either settings (M10 or M20), both types will be received.

Currently there is a minor issue in the code: M10 configuration will show as M20 if no signal is being received. I can fix that easily in the next version. So it will show the configured type (but as soon as a signal is received, the real type, not the configured type, will be shown)

F5MVO, 2021-09-28:
thanks for the quick answer.