dl9rdz/rdz_ttgo_sonde #198: livemap name error

14ri004, 2021-09-29:
when I scan 2 probe frequencies, the name remains the same, I have Aberporth (uk) and Bourges (fr), this displays 2x Bourges in the Livemap (see screenshots)

[image]
[image]

eben80, 2021-09-30:
I'm curious,
What is the expected/previous behavior? I haven't experienced this use-case before.

dl9rdz, 2021-09-30:
I guess the live map is not of much use in such a case where no signal is received and the TTGO is in scan mode...

But anyway, if you want my opinion on what might be the expected behaviour:
When the scan mode switches to a frequency on which no prior signal was received, show launch site, freq, mode, but no other info.
When the scan mode switches to a frequency on which a signal was received before: show details of the signal. So for 405.6 MHz I would expect "S1710339" instead of "Bourges", and for 404.6 MHz, I would expect all the data from the DFM sonde.

Possibly, implementing this is as easy as this change to livemap.js (but I did not test it, maybe more needs to be done):
```
-  if (data.id && data.vframe != lastframe ) {
+  if (data.id) {
```

On the map, maybe it would be nice to keep all sonde items instead of just the most recently received on.

eben80, 2021-09-30:
I remember the reason we had 
`  if (data.id && data.vframe != lastframe ) {`
was to retain the last received rssi level when the signal was lost temporarily.
I'm just wondering what receiving two sondes  simultaneously would look like on LiveMap?
Do we want a marker, track and prediction for each? The info for which sonde would we display in the sonde detail section?

14ri004, 2021-09-30:
good evening, I was decoding the probe, and I had just lost the signal momentarily, and during this time the ttgo resumed its scan, normally the names should have been Aberporth and Bourges, but he put Bourges on the 2 frequencies, it does not prevent it from working but it is strange, thanks

closed 2021-10-01 by 14ri004