dl9rdz/rdz_ttgo_sonde #205: Meteomodem M10 Battery Voltage & Humidity

LukePrior, 2021-10-02:
We now have battery voltage decoding for RS41 and DFM so I think it would also be good to have for M10. The M10 is the 3rd most popular sonde decoded by this software after RS41, and DFM. The code for battery voltage decoding in auto rx can be found here: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/demod/mod/m10mod.c#L828

The addition of Humidity decoding would bring this in line with auto rx but given the complexity and scope, you may wish to exclude it. The code can be found here if interested: https://github.com/projecthorus/radiosonde_auto_rx/blob/master/demod/mod/m10mod.c#L774

dl9rdz, 2021-10-02:
Should be ok

LukePrior, 2021-10-03:
Yep it's all perfect,

closed 2021-10-03 by LukePrior