dl9rdz/rdz_ttgo_sonde #206: Wont compille...

kmp56, 2021-10-02:
Wont compille in Arduino IDE with _#define FEATURE_SONDEHUB 1_
 in _features.h_ . When set to 0 is ok.

dl9rdz, 2021-10-02:
TravisCI disagrees.
Image on rdzsonde.mooo.com was automatically compiled with Arduino IDE, with FEATURE_SONDEHUB set to 1.

Feel free to share the error message you are getting.

kmp56, 2021-10-02:
RX_FSK.ino stops on line:  _shImportInterval = 5;   // refresh now in 5 seconds_
console error description: _RX_FSK.ino: In function 'void setupConfigData()':
RX_FSK:595:3: error: 'shImportInterval' was not declared in this scope
   shImportInterval = 5;   // refresh now in 5 seconds
   ^_
_'shImportInterval' was not declared in this scope_

dl9rdz, 2021-10-02:
So I guess you wanted to say it wont compile with FEATURE_SONDEHUB 0
When set to 1 t is ok?
That is something I can reproduce and fix...

kmp56, 2021-10-02:
YES, When set to 1 is OK !

kmp56, 2021-10-02:
Sorry, my error in first post...
Wont compille in Arduino IDE with #define FEATURE_SONDEHUB 0
in features.h . When set to 1 is ok.

kmp56, 2021-10-02:
in ver. 00899db is OK
THANKS ! GREAT PROJECT !

closed 2021-10-02 by kmp56

LukePrior, 2021-11-04:
@kmp56 are you running a fork of rdzTTGOsonde named RadioSONDY on version 01.11.2021?

kmp56, 2021-11-04:
W dniu 04.11.2021 o 10:51, Luke Prior pisze:

> @kmp56 <https://github.com/kmp56> are you running a fork of 
> rdzTTGOsonde named RadioSONDY on version 01.11.2021?
> 
> or Android 

Yes. I am experimenting with the Polish version of rdzTTGOsonde. The 
most important screens are in Polish, the mDNS name and password are 
changed. Also the web configuration page has been partially translated.

kmp56

LukePrior, 2021-11-04:
> Yes. I am experimenting with the Polish version of rdzTTGOsonde. The most important screens are in Polish, the mDNS name and password are changed. Also the web configuration page has been partially translated.

Ahh ok that explains it, I just saw the data in the SondeHub database and was curious.