dl9rdz/rdz_ttgo_sonde #208: Unable to connect to MQTT

ZL1LAC, 2021-10-03:
I can seem to get my ttgov2 to connect to the MQTT server.  I get the following error.

``Client <unknown> disconnected due to protocol error``

The username and password are correct, I can connect to the server using MQTT explorer just fine, what am I missing?

dl9rdz, 2021-10-03:
Thanks, yes, this was broken (has always been, surprising that it did not cause more problems before).

ZL1LAC, 2021-10-03:
Hi, I just fixed it for myself.

I changed the ``MQTT client ID`` value from rs41 to just 41, rebooted, and now it's happy.

[image]

dl9rdz, 2021-10-03:
Well, not really, the bug is that the mqtt placed the dynamically generated client id (from the config, but appending a random number) on the stack.  This value can and will be overwritten later while the mqtt library is using it. So basically it is rather unpredictable if it will work or not. I would say the observation that it worked now for you is not due to the id change, but to luck.

Anyway, I just commited a fix for that issue, so now it should be fine.

ZL1LAC, 2021-10-03:
Oh alright, when the fix be live?  I got an expected sonde launch in about an hour.

dl9rdz, 2021-10-03:
It already is. Since 10 minutes ago.

ZL1LAC, 2021-10-03:
Ah cool so update my ttgov2 LoRa then?

dl9rdz, 2021-10-03:
Yes, should be good to update.

closed 2021-10-03 by ZL1LAC