dl9rdz/rdz_ttgo_sonde #214: Sondehub Import Frequency rounding

eben80, 2021-10-08:
Hi,
I've enabled the Sondehub frequency import and it works. 
It looks however that it might need to round the frequency a bit. I guess this might be in issue with the frequency coming from Sondehub.

[image]

LukePrior, 2021-10-08:
This is unrelated but we should make the default list of frequency/station/type combinations shorter so that there are more available positions for SondeHub by default.

eben80, 2021-10-10:
Thanks for that very thorough response. Yes I'm not getting the sonde decoded on the sondehub provided frequency even.  So in the scenario from the screenshot, only the 400.6MHz had any output on the data page.  As you would expect if the frequency IS actually 400.6

Too bad that the configured frequency is not being sent as part of telemetry 😄 

I like your second approach on the TTGO but then, as you say, if the auto_rx frequency is out of calibration, I wouldn't be able to tune to the sonde on the frequency sent by sondehub anyway so I will close it on here. It seems the issue needs to be solved with auto_rx calibration.

closed 2021-10-10 by eben80

dl9rdz, 2021-10-10:
> Too bad that the configured frequency is not being sent as part of telemetry 😄
It actually is sent for RS41 (not for others).
No idea why some autorx stations are not sending the right frequency from decoded RS41 telemetry though....

eben80, 2021-10-10:
Oh you're right, I missed this:  

> 0x010: Frequency/Firmware