dl9rdz/rdz_ttgo_sonde #238: Update SondeHub Import To Use TX Frequency if available?

LukePrior, 2021-10-30:
To address concerns with some auto rx stations uploading bad frequency information which was causing errors when we used the /sondes API to automatically import frequencies we now also upload the transmitted frequency from RS41, and RS92 sondes. This field is optional and won't be present in all data so we will need to fall back to frequency if it is not available.

The new field `tx_frequency` should probably take precedence over `frequency` if both are present. Here is an example request with both fields present: https://api.v2.sondehub.org/sondes/telemetry?duration=0&serial=T2410481&datetime=2021-10-30T06:54:53.001000Z

You can see that `tx_frequency` is `405.3` while `frequency` is `405.301`. Do you think it would be a good idea to take advantage of this?

dl9rdz, 2021-10-30:
Ok, latest version (devel20211030) should use that (but it is untested so far)
TravisCI is not running right now (no idea why), so no new binary image yet, just the source code in the repo.

LukePrior, 2021-10-30:
Will close once someone can confirm this is working.

eben80, 2021-10-31:
Hi,
On the lunchtime flight it automatically selected  one of my deselected entries on the correct frequency. I wasn't able to watch was reported on SondeHub at the time.

I guess it will not automatically deselect it after it disappears from the api response right?

dl9rdz, 2021-10-31:
Yes that is right. Currently it is activated, but not deactivated automatically. Maybe something to enhance in the future

eben80, 2021-10-31:
Great thanks. BTW, for the first time now I've had to actually have the no_ota partition scheme to also have my small modifications included. Storage is getting tight 😁

dl9rdz, 2021-10-31:
Yes, not much space left...

I am hoping for https://github.com/me-no-dev/arduino-esp32fs-plugin/pull/34 to get accepted. Then fonts can be moved to a new partition outside the two OTA partitions (cut off the spiffs partition), and still everything remains operational, including OTA updates and file upload using Arduino IDE.  That would even give enough space for latin1 fonts instead of 7-bit ascii for the TFT display...

eben80, 2021-10-31:
Crossing fingers for that then 🤞 That sounds like it would bring some relief.

TimMcMahon, 2021-11-01:
Config screen shows the navbar but nothing below it.

devel202111101

LukePrior, 2022-01-11:
I think it is safe to assume that this is working as expected.

closed 2022-01-11 by LukePrior