dl9rdz/rdz_ttgo_sonde #244: DFM sondes need validity checks regarding their position

dm5tt, 2021-11-04:
**Describe the bug**

I'm observing a lot of DFM radiosondes from the Calw military area. From time to time there seem to be some position glitches.

**To Reproduce**
Receive a DFM radiosonde  with low SNR (-99dBm to -102dBm).

**Expected behavior**
Filtering out invalid position data based on the last valid ones.

**Screenshots**

[image]

**Version and specific configuration:**
 - Version: devel20211101
 - Setup: 2m/70cm mobile antenna -> LNA -> 403MHz SAW BPF -> TTGO
 - 
**Additional context**
Add any other context about the problem here.

LukePrior, 2021-11-04:
Also there are just more validity checks that can be done on DFM sondes, I believe this is planned for the next master version.

TarmoKolli, 2021-11-16:
I have the same problem with DMF probes. Unfortunately, my antenna only sees the eastern sky, but a few tens of kilometers away from me is a military area from which the signal from the probes fired is distorted. Therefore, my receiver sends the last probe information to the server with the penultimate probe serial number.
[image]

dl9rdz, 2022-01-24:
There should be some enhancements in the latest version.

Serial numbers for DFM are always cleared after 30 Minutes. They are also cleared, if a position deviates substantially from the previously decoded position. Let me know if that fixes the issue,

TarmoKolli, 2022-01-25:
I'll let you know as soon as possible. Unfortunately, this can take time, as the DFM-09 probes are used by the military, sometimes during exercises.

TarmoKolli, 2022-02-10:
Two DFM09 sondes were sent today and the picture is better, but unfortunately my receiver has sent information about one non-existent 60019549 sonde.
[image]

closed 2026-11-10 by dl9rdz