dl9rdz/rdz_ttgo_sonde #245: Battery info screen

pavolgaj, 2021-11-07:
I don't know if it's only a problem with my ttgo or battery but I can see only a value of battery voltage on the battery info screen. Battery status, temperature, USB voltage are empty.
I have ttgo lora32 with a connected LiPo battery and use screen1.txt file.

dl9rdz, 2021-11-07:
The full information is available on on boards with a AXP192 battery controller (T-Beam , M5Stack should also work).
Other boards such as the ttgo lora32 v2.1 have only a voltage divider across the battery to an adc input, which allows measuring the battery voltage, but nothing more.

pavolgaj, 2021-11-07:
Thanks. It would be nice to have some table (maybe on wiki page) with available options for different ttgo types.
It would be useful to show also a graph with battery voltage - as information for controlling charging. But, its implementation could be too complicated...

eben80, 2021-11-08:
> Thanks. It would be nice to have some table (maybe on wiki page) with available options for different ttgo types. It would be useful to show also a graph with battery voltage - as information for controlling charging. But, its implementation could be too complicated...

With the Lora32 v2.1 board you don't have to worry about overcharging(ones with proper components), you just have to take car that you don't over-discharge. This you can do by keeping an eye on the battery voltage of course.