dl9rdz/rdz_ttgo_sonde #261: Unable to send data to radiosondy.info

TarmoKolli, 2021-12-24:
Unable to send data to radiosondy.info.
 However, the data will reach Sondehub.org

Unfortunately, I don't have an amateur radio call, I have a username and password for radiosondy.info
Somewhere in the password shows 0 i, although I've tried to enter it.

[image]
.

dl9rdz, 2021-12-24:
radiosondy uses the standard aprs-is authentication method. You need a passcode.

Originally, this passcode was meant to restrict access to radio amateurs only. However, it is a very week form of authentication, and passcode generators can easily be found on the Internet (just google for aprs passcode).

Nevertheless, I suggest you read the usage conditions of aprs-is first (http://www.aprs-is.net/), and only forward data there if you comply with those conditions.

As you are not a licensed radio amateur, I suggest you use radiosondy's port 14590 (non-forwarding) instead of port 14580 (forwarding to the aprs-is network)

TarmoKolli, 2021-12-24:
Thank you. Looks like it works now. Merry Christmas to you!
[image]

TarmoKolli, 2021-12-26:
Now I got it - previously the data was sent, but it was not reflected in the user's statistics. 
I had to change the username a bit - I looked at the "https://radiosondy.info/user/user_panel.php" page. 
On the "Raspberry Pi System Monitoring" section of this page, I found the name of my receiver, to which I added the sequence number -11. 
Now everything seems to be working.
[image]

closed 2021-12-26 by TarmoKolli