dl9rdz/rdz_ttgo_sonde #262: Probe frequency offset

14ri004, 2022-01-01:
Hello, would it be possible to add the offset of the frequency of the probe to the Livemap, as in the photo of the ttgo screen,
[image]

14ri004, 2022-02-17:
Hello, I use 2 TTGOs which are at a distance from my house, so I cannot see the offset of the probe offset on the screen, would it be possible to add this parameter to the Livemap, this which will allow me to be able to calibrate the frequency without having to move every time, thank you

dl9rdz, 2022-02-17:
I will keep that on my todo list.

For now, what you can do (if you have network connectivity to your TTGO) is to connect port 14570. This is the port used by the Android app. You will get JSON data that includes the afc value.

g8ixn, 2022-02-25:
I have found that it is nessasary to reboot (power off )  after adjusting offset.
Maybe should be noted in the line after  > RX frequency offset (Hz)
Thanks

dl9rdz, 2022-02-25:
No it is not.  A modified offset is used upon the next frequency change. 

This means that it is not automatically used while continuously eceiving a sonde, but as soon as you change frequency or go back to the scanner, the new offset is used.

g8ixn, 2022-02-25:
Thanks for reply.I only receive one frequency mostly.Will put an unlikely one in memory 2 & tick both as a workaround.