dl9rdz/rdz_ttgo_sonde #265: Automatically scan for sondes?

LukePrior, 2022-01-10:
I'm assuming that this isn't possible with the ESP32 hardware but the ability to automatically scan the 400 - 406 MHz spectrum and find any transmitting sondes without having to know the frequencies in advance would be extremely useful.

73-de-LZ, 2022-02-06:
Extremely for what ? If you leave in area, where there is no other automatic stations, or waiting for military balloon launching at unknown frequency?
In fact if scanning is not on known frequencies it will take more time and you will loose data when sondes comes in range.
Scanning on known frequencies is faster, that's why you can configure only QRG of interest for you or your area.

dl9rdz, 2022-02-06:
Sequential scanning with decoders would be very slow. For example, full 6 MHz in 5 kHz steps, 3sec on each freq => 1200*3 seconds or 60 Minutes.  And that's a single sonde type.

I am considering this for the spectrum scanner. This one already has a peak detector, and it could try to decoder on the detected peak. But that would work only for rather strong signals, visible in the fast spectrum scan, i.e. signals a few dB above the decoding limit. And, in its simplest form, only for one (the strongest) signal on the band.

TimMcMahon, 2022-02-07:
Radiosondes are launched from two sites, one a training facility, nearby and I've observed them use the following frequencies:
| Frequency MHz |
| 401.5 |

TimMcMahon, 2022-02-07:
Radiosondes are launched from two sites, one a training facility, nearby and I've observed them use the following frequencies:
| Frequency |
| -------------: |
| 400.5 MHz |
| 400.8 MHz |
| 400.9 MHz |
| 401.2 MHz |
| 401.4 MHz |
| **401.5 MHz** |
| 401.8 MHz |
| 402.0 MHz |
| 402.3 MHz |
| 402.5 MHz |
| 402.6 MHz |

The training facility might launch half a dozen radiosondes in one day.

Usually if a radiosonde fails, another is launched shortly after with a different frequency.

TarmoKolli, 2022-02-07:
My list of frequencies. Frequencies within a radius of 250 km used in recent years. The two basic frequencies are double
[image]
.