dl9rdz/rdz_ttgo_sonde #268: Show error message if SondeHub position uploading enabled with default callsign or location

LukePrior, 2022-01-13:
We should show a message in the website interface if the user has enabled SondeHub position uploading but hasn't changed the default callsign of CHANGEME_RDZTTGO or has no position or 0,0 position set.

I'm not sure if this is also possible but it would be good to display any error messages when trying to upload data in the website interface also.

DL2MF, 2022-01-21:
You should better use common unrouted callsigns, like implemented in several HAM/APRS services.
Look for NOCALL / N0CALL 
[(https://github.com/hessu/perl-aprs-fap/)]

dl9rdz, 2022-01-21:
This "callsign" is specifically in the sondehub configuration and will only be used in the communication with sondehub, which does not forward or routed anywhere else anyway...

I have considered more status information somewhere on the web interface, but so far its low priority on my TODO list. But I consider it a potential extension (just need to be careful, not much code space left...) I am still hoping for https://github.com/me-no-dev/arduino-esp32fs-plugin/pull/34 to be merged, then I can change the memory layout and have more code space without breaking usability with the Arduino IDE...

LukePrior, 2022-01-22:
Can you just run a custom branch of Arduino ESP32 filesystem uploader with your changes as the current one hasn't been updated in years?

dl9rdz, 2026-11-10:
Old issue, probably fine to close (and SondeHub status responses including error message are shown on the web status page now for some time)

closed 2026-11-10 by dl9rdz