dl9rdz/rdz_ttgo_sonde #274: [feature request] Add distance to MQTT report / minimum distance

g3gg0, 2022-02-03:
Hi, great piece of software, thanks a lot.

I am using MQTT also for push information for my home stuff.

Could you please add the calculated distance to the report sent via MQTT?
It would also be really helpful to set a minimum distance at which MQTT packets get sent.
So whenever a sonde enters e.g. a 25km bubble, my mobile will display the message.

Thanks

dl9rdz, 2022-02-03:
Interesting idea, but not sure if it is the right thing to implement on the TTGO (where code space in scarce).

Would it not be easier to have such functionality implemented elsewhere in the network, like in a simple client that subscribes to the MQTT messages and creates alerts based on distance?

DL2MF, 2022-02-12:
Hi, 
I agree to DL9RDZ and ressoucres on TTGO are very rare. To read distance from MQTT is about the same effort than additionally calculate the current distance from actual position (and use it for other nice things like a messenger bot for notification),

On your client side you can simply subscribe to the messages like in my JS example:

function onConnect() {
    client.subscribe('rdz_sonde_server/packet');
};

The parse the message by a second function on each message arriving:

function onMessageArrived(message) {
	const m = JSON.parse(message.payloadString)
...
}

Now you can access m.lat and m.lon objects to use these as parameters for your distance calculation.