dl9rdz/rdz_ttgo_sonde #275: [feature request] pls Add ddns function to webserver

laco717, 2022-02-04:


DL2MF, 2022-02-12:
Why? This will be another waste of ressources and you might notice, that an ESP32 is not secure and reliable enough, to operate it 24/7 exposed to the web.

cixio, 2022-02-14:
I don't think the idea is that bad, especially if you operate the ttgo in different networks on the same computer / smartphone.

I use it at home in my wirless network and on the go - both with my smartphone. It is easier to have one bookmark (which will work always instead of two or always enter the (maybe changing) ip-adress.

A ddns is not needed to get it working exposed to the web... but sure, THIS is really not secure.

yaleman, 2022-02-14:
> Sorry, mixed up mdns / **dynaminc** dns.

mDNS discovery's already there, too 😄  (rdzsonde.local)